euslisp/EusLisp #79: Could not open piped fork stream more the 20k

k-okada, 2015-01-04:
Reported by @Kindsenior

It seems there are limitation in numbers if piped fork streams
```
(setq *p* (piped-fork "bash"))
(progn (setq i 0) (do-until-key (format *p* "ls~%") (warn " ~s" i) (incf i)))
```

I doubt that it forget to close file stream.

garaemon, 2015-01-04:
PIPE has limited buffer size on POSIX system (see [this topic](http://stackoverflow.com/questions/4624071/pipe-buffer-size-is-4k-or-64k)).
If write buffer (child process -> parent process) exceeds the limit, the child process will hang at the point of writing something to the buffer.

So, in order to avoid this problem, what we can do is just call `(read *p*)`. It's not a bug of euslisp or POSIX.

garaemon, 2015-01-05:
I think it's ok to close this issue.

closed 2015-01-05 by k-okada