euslisp/EusLisp #82: Release euslisp/jskeus as Homebrew recipe for OSX

furushchev, 2015-02-04:
Because of recent change of euslisp package, maybe it seems that euslisp must be installed under non ROS package system.(e.g. apt-get for Ubuntu)
On the other side, ROS supports Homebrew and has own tap(tap is similar to ppa on apt system).
Homebrew also has binary distribution system called Bottle.
So I intend to release euslisp (and jskeus) as homebrew packages.

1. Maybe need to establish test system of this project on travis.
2. we need to choose tap destination (homebrew original or ros specified tap)

How do you think of this? (Is there any problem?)

k-okada, 2015-02-04:
who going to maintain that release and how long?

furushchev, 2015-02-04:
I think recipe file can be autogenerated as it is first scripted on the timing of tagging github.
It is just simple script file like: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/graphviz.rb

k-okada, 2015-02-04:
So do you mean it so easy that I should maintain ?

furushchev, 2015-02-04:
Need not to maintain, I think., I'll write recipe file. You mean it? (Sorry I cannot make out very much...)
In the first place, euslisp has OSX specific Makefile, if this repo supports OSX, I take count that this package must also be maintained...

k-okada, 2015-02-04:
We need someone who  answer "I can not compile euslisp on my mac", "I
cannot download tap, what should I do?", can you answer 12 month later? 24
month later?

furushchev, 2015-02-04:
Still I cannot make out...
"I can not compile euslisp on my mac" -> This should be maintained whether released on homebrew or not, because this package seems to already supports OSX.
"I cannot download tap, what should I do?" -> it depends on homebrew and ros community. (ROS has own homebrew tap), so I think no worry about it.
So you mean that OSX user should install only by downloading and executing makefile?

garaemon, 2015-02-04:
I think it's a good start point to add os x build test to travis (as allow failure?).

k-okada, 2015-02-05:
> I think it's a good start point to add os x build test to travis (as allow
> failure?).—
>
>
> Yes ,I agree. please do that.

 > "I can not compile euslisp on my mac" -> This should be maintained
whether released on homebrew or not, because this package seems to already
supports OSX.
If you can take care of this kind of question for at least next 60 month,
it ok. we're happy to announce that euslisp is now ready for mac, and I
hope you do that.

> "I cannot download tap, what should I do?" -> it depends on homebrew and
ros community. (ROS has own homebrew tap), so I think no worry about it.
OK, question is  who will answer question like "I can't donwnload euslisp
from tap, what can I do" and again, I hope you do that next 60 month.

> So you mean that OSX user should install only by downloading and
executing makefile?
I mean current maintainer is not using OSX so if you are going to release
euslisp package on new platform, we have to find someone who can take care
of that.

wkentaro, 2015-10-16:
I sent PR to homebrew at https://github.com/Homebrew/homebrew/pull/45034, and actually this is suggested by @k-okada .

For homebrew, we can set version to release, so I think we **don't need take much time to maintain it**.
Because at the version [9.15](https://github.com/euslisp/EusLisp/releases/tag/EusLisp-9.15) **euslisp can be compiled on OS X**, and **if we do not upgrade the version which is released** on Homebrew, 
we don't need to concern about the error while compilation.

One concern is the time when OS X is upgraded, and at that time euslisp can be not compilable on that OS X version.
At that time, we can select whether we fix it and release new version or remove euslisp from Homebrew's official formula list.

bfontaine, 2015-10-16:
> or remove euslisp from Homebrew's official formula list.

We don’t remove formulae from Homebrew; we move them to a special repository for broken formulae ([homebrew/boneyard](https://github.com/Homebrew/homebrew-boneyard)). However you can maintain the formula in your own tap ([see the doc here](https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/How-to-Create-and-Maintain-a-Tap.md)) which allows you to do pretty much what you want with it.

wkentaro, 2015-10-16:
@bfontaine 
Thanks for your nice information. I have ever created tap [here](https://github.com/wkentaro/homebrew-trr) and it was merged to official repository.

wkentaro, 2015-10-16:
formula for jskeus is [here](https://github.com/Homebrew/homebrew-x11/pull/143)

wkentaro, 2015-10-17:
To be merged in homebrew, I need to resolve below syntax error:

```
* Non-libraries were installed to "/usr/local/Cellar/euslisp/9.15/lib"
Installing non-libraries to "lib" is discouraged.
The offending files are:
  /usr/local/Cellar/euslisp/9.15/lib/eus.init.l
  /usr/local/Cellar/euslisp/9.15/lib/eusglrt.l
  /usr/local/Cellar/euslisp/9.15/lib/EUSRC
  /usr/local/Cellar/euslisp/9.15/lib/eusrt.l
```

It says installing the [lib](https://github.com/euslisp/EusLisp/tree/master/lib) files which are not libraries is not recommended.
I was told by one of the maintainers to describe how much difficult to rename the directory name (not by me but by Euslisp's maintainers).
https://github.com/Homebrew/homebrew-x11/pull/142#issuecomment-148881708

Could anyone answer for it?

k-okada, 2015-10-18:
> Homebrew/homebrew-x11#142 (comment)

where is the EUSDIR in homebrew? euslisp/9.15 ? if so, that's make sense to
install them to euslisp/9.15/lib (
https://github.com/euslisp/EusLisp/blob/74fff34afa261db392800f1f2ea7518ee4cadfff/lisp/c/eus.c#L1133
)
if you can set EUSDIR to different directly such as
share/euslisp/jskeus/eus as ROS package deos. then we can avoid this problem

wkentaro, 2015-10-18:
> where is the EUSDIR in homebrew? 

Yes, `EUSDIR` is set as `/usr/local/Cellar/euslisp/9.15`.

> if you can set EUSDIR to different directly such as
share/euslisp/jskeus/eus as ROS package deos.

I can try it.
Actually I think jskeus and euslisp should be released as different package with dependency of jskeus on euslisp, so the construction of directory will be different from that in ROS>

wkentaro, 2015-11-30:
done at https://github.com/Homebrew/homebrew-x11/pull/143.
So maybe you can close this issue.

furushchev, 2015-11-30:
@wkentaro Thanks for working.

closed 2015-11-30 by furushchev

k-okada, 2015-11-30:
- please add section 1.2 to
https://github.com/euslisp/jskeus/blob/master/README.md
- can we run travis test ? https://github.com/euslisp/jskeus/pull/302
- if we can automate release process, that would be nice.

wkentaro, 2015-11-30:
> - if we can automate release process, that would be nice.

How do you release currently?

k-okada, 2015-11-30:
if you integrate this with catkin and ros build firm process, that 's nice,
otherwise, we'll add command on travis,

wkentaro, 2015-11-30:
What commands do you run when releasing?