euslisp/EusLisp #83: Feature request, introduce EUSLISP_LOAD_PATH environment variable like PYTHONPATH

garaemon, 2015-02-08:
Feature request of EUSLISP_LOAD_PATH.
https://github.com/euslisp/EusLisp/blob/master/lisp/l/loader.l#L173

Currently `*load-path*` is defined by EUSDIR, but it would be useful we can use EUSLISP_LOAD_PATH environment variable like PYTHONPATH in order to use `require` without path to lisp file.

furushchev, 2015-02-08:
+1 it is quite useful

k-okada, 2015-02-08:
what is the use case? recent programmer uses euslisp on ros and they can
use pcakge://

garaemon, 2015-02-08:
We can remove the "package://".

before
```lisp
(require :pr2 "package://pr2eus/pr2.l)
```

after
```lisp
(require :pr2)
```

It means that we do not need to check "package://pr2eus/pr2.l" or "package://pr2eus/euslisp/pr2.l".

And also I think we can handle compiled lisp code better because we do not need to specify path to lisp code.

k-okada, 2015-02-09:
so are you proposing
1) EUSLISP_LOAD_PATH
and
2) require searches not only load path and  also subdirectory of the load
path

as for 1), we can update roseus.l to set *load-path* from
genev("ROS_PACKAGE_PATH") but not sure if it is useful, for 2)  we can
extend require to (requrie :pr2eus "pr2.l") or something, search pr2.l from
all ROS_PACKAG_PATH may dangerous.

garaemon, 2015-02-10:
My proposal is only 1) and I don't think including all the subdirectory to search path. But including "euslisp" subdirectory may be reasonable.

k-okada, 2015-10-19:
some discussion on https://github.com/jsk-ros-pkg/euslib/pull/1#issuecomment-136678383

wkentaro, 2015-10-19:
I have done an implementation at https://github.com/jsk-ros-pkg/jsk_roseus/issues/379 for the similar feature discussed on this issue. 
The syntax is
~~~
catkin_eus_setup(
 PACKAGE_DIR .
 )
 ~~~
 In CMakeLists.txt
 ~~~
 (require :pr2eus/pr2)
 ~~~
 To include other files in package.

wkentaro, 2015-10-19:
> My proposal is only 1) and I don't think including all the subdirectory to search path. But including "euslisp" subdirectory may be reasonable.

My implementation does not assume the path "euslisp", because we can specify it with `catkin_eus_setup(PACKAGE_DIR)`.