euslisp/EusLisp #84: git clone is heavy

garaemon, 2015-02-09:
It take 14 minutes
```
$ time git clone https://github.com/euslisp/EusLisp.git
Cloning into 'EusLisp'...
remote: Counting objects: 16146, done.
remote: Compressing objects: 100% (3874/3874), done.
remote: Total 16146 (delta 12530), reused 15674 (delta 12191)
Receiving objects: 100% (16146/16146), 40.18 MiB | 52 KiB/s, done.
Resolving deltas: 100% (12530/12530), done.
hub clone https://github.com/euslisp/EusLisp.git  4.16s user 0.91s system 0% cpu 13:55.06 total
```

k-okada, 2015-02-09:
remove old files under model and doc, 
```
(cd EusLisp.org; git pull origin master)
rm -fr EusLisp
cp -r EusLisp.org EusLisp
du -sh EusLisp/.git
java -jar ~/Downloads/bfg-1.12.1.jar --delete-folders models EusLisp
java -jar ~/Downloads/bfg-1.12.1.jar --delete-folders html EusLisp
java -jar ~/Downloads/bfg-1.12.1.jar --delete-files 'manual.{pdf,dvi,ps,idx}' EusLisp
java -jar ~/Downloads/bfg-1.12.1.jar --delete-files 'jmanual.{pdf,dvi,ps,idx}' EusLisp
(cd EusLisp;git reflog expire --expire=now --all && git gc --prune=now --aggressive)
(cd EusLisp;git verify-pack -v .git/objects/pack/*.idx | sort -k 3 -n | tail -200 | awk '{"git rev-list --objects --all|grep " $1 | getline fname; print fname" "$2" "$3" "$4}'|column -t  | sort -k 2)
du -sh EusLisp/.git
```
.git becomes 41M to 18M

garaemon, 2015-02-24:
heavy again?
git clone takes 28 minutes and the size of directory is 106M

garaemon, 2015-02-24:
```
  garaemon  @ueda-t430:  Yes, Master?  
$ time git clone https://github.com/euslisp/euslisp.git
Cloning into 'euslisp'...
remote: Counting objects: 17765, done.
remote: Compressing objects: 100% (570/570), done.
remote: Total 17765 (delta 243), reused 18 (delta 18), pack-reused 17177
Receiving objects: 100% (17765/17765), 42.95 MiB | 49 KiB/s, done.
Resolving deltas: 100% (13306/13306), done.
hub clone https://github.com/euslisp/euslisp.git  4.74s user 1.21s system 0% cpu
 28:31.66 total
$ du -h . -d 0
106M    .

```

k-okada, 2015-02-24:
it is 7 sec for me,
```
k-okada@kokada-t440s:~$ cd /tmp/
k-okada@kokada-t440s:/tmp$ git clone https://github.com/euslisp/euslisp
Cloning into 'euslisp'...
remote: Counting objects: 17765, done.
remote: Compressing objects: 100% (570/570), done.
remote: Total 17765 (delta 243), reused 18 (delta 18), pack-reused 17177
Receiving objects: 100% (17765/17765), 42.95 MiB | 11.24 MiB/s, done.
Resolving deltas: 100% (13306/13306), done.
Checking connectivity... done.
k-okada@kokada-t440s:/tmp$ git clone https://github.com/euslisp/euslisp
fatal: destination path 'euslisp' already exists and is not an empty
directory.
k-okada@kokada-t440s:/tmp$ rm -fr euslisp/
k-okada@kokada-t440s:/tmp$ git clone https://github.com/euslisp/euslisp
Cloning into 'euslisp'...
remote: Counting objects: 17765, done.
remote: Compressing objects: 100% (570/570), done.
remote: Total 17765 (delta 243), reused 18 (delta 18), pack-reused 17177
Receiving objects: 100% (17765/17765), 42.95 MiB | 10.62 MiB/s, done.
Resolving deltas: 100% (13306/13306), done.
Checking connectivity... done.
k-okada@kokada-t440s:/tmp$ time git clone https://github.com/euslisp/euslisp
fatal: destination path 'euslisp' already exists and is not an empty
directory.

real 0m0.003s
user 0m0.000s
sys 0m0.001s
k-okada@kokada-t440s:/tmp$ rm -fr euslisp/
k-okada@kokada-t440s:/tmp$ time git clone https://github.com/euslisp/euslisp
Cloning into 'euslisp'...
remote: Counting objects: 17765, done.
remote: Compressing objects: 100% (570/570), done.
remote: Total 17765 (delta 243), reused 18 (delta 18), pack-reused 17177
Receiving objects: 100% (17765/17765), 42.95 MiB | 11.48 MiB/s, done.
Resolving deltas: 100% (13306/13306), done.
Checking connectivity... done.

real 0m7.928s
user 0m4.891s
sys 0m0.571s

```

garaemon, 2015-02-24:
hmm, how much is size of euslisp?

✉︎ Ryohei Ueda