euslisp/EusLisp #85: [Feature request] add --quiet option

garaemon, 2015-02-12:
I see this topic several times and now I have it again...

I'd like to launch euslisp without bootstraping messages.
I missed several important error message because of a lot of bootstraping messages from multiple euslisp processes.

furushchev, 2015-02-12:
+1 I'd rather prefer --verbose or --debug option than --quiet

garaemon, 2015-02-12:
> +1 I'd rather prefer --verbose or --debug option than --quiet

I think we should not change default behavior and the information of bootstrap messages are sometimes 
important for debugging.

For example, one student says "It does not work blah blah blah" and expert can know error from the boostrap message.

And now we gradually try to use deb package, and version of euslisp will be more important.

k-okada, 2015-02-12:
the original behavior (for me, so it is just a 5 yeas ago) was to print verbose info to stdout and use redirect to dev/null to not to show verbose messages. for example
https://github.com/jsk-ros-pkg/jsk_roseus/blob/1.1.23/geneus/scripts/genmanifest_eus#L11
assume this behavior, but at some point we send some message to stderr to invoke euslisp within piped-fork.

if we use `output=log` and bootstrap message goes to stdout, then I think everyone is happy.

```
piped-forkは，Linuxの方でちゃんと動いている？
 
cygwinでは，下のように，標準出力に表示されてしまっているものを読み飛ばして
プロンプトまで表示するようになっているので，そこまで読み飛ばせば値が返るけれど
linuxの方は，最初の値も返ってこなくない？
 
piped-forkされたeusの場合には，is-ttyだったか，標準入力がキーボードかソケットポート
かを調べてプロンプトを出す出さないを決めていたけれど，meadowかemacsのバッファ内
でキーボード扱いされないということで，プロンプトを出すようにしていた結果，
piped-forkで立てられたプロセスはプロンプトまで読み飛ばすという形になっていますね．
SunOS, Solarisの時代には，現在モジュール名を標準出力へ出しているような形はとら無い形
だったので，formatで送ってすぐにreadすれば計算結果を得られるという形でした．
 
piped-forkでeusの別プロセスを立ち上げる場合には，
ユーザのeusアプリでも同じように，標準出力への表示は行わず，標準エラー出力ポートへ
出すようにしておけば計算結果だけを出すようになってよくなるかな．
セミコロンでコメントだとわかる出力であれば問題ないからそれでもいいはずですが．
とにかく，eus readでエラーが起こらない書式の表示であればプロンプトまで読み飛ばす形は
悪くないのかもしれないけども，linuxではどうなっているかを確かめようとして
何も返ってこないような感じだったので私の環境だけかどうか確認したい．
 
ちなみに，オブジェクトを読み出すことになる，たとえば，
(format a "(make-cube 10 20 30)~%")
の結果を受け取るためには，
あらかじめ
(format a "(setq *print-circle* t *print-object* t)~%")
が必要です．
＃<body #Xxxxxxx  (xxxx) >の表示ではなく
内容を表示するための*print-object*をtにして，cubeの中の再帰データ構造を
無限に表示しないようにするための*print-circle*の大域変数をtにしておく
機構で，知らない人も多いかと思います．シリアライズするためのCommonLispに
ある機構です．
 
eusgl$ (setq a (piped-fork "eusgl"))
#<io-stream #X1010a488>
eusgl$ connected to Xserver DISPLAY=localhost:0.0
X events are being asynchronously monitored.
�[34m
EusLisp 8.26 for Cygwin  created on Fri Jun 19 09:20:27 2009�[0m
nil
eusgl$ (format a "(setq *print-circle* t *print-object* t)~%")
nil
eusgl$ (read a)
readmacro
eusgl$ (until (eq 'eusgl$ (read a)))
nil
eusgl$ (read a)
t
eusgl$ (read a)
eusgl$
eusgl$ (format a "(make-cube 10 20 30)~%")
nil
eusgl$ (read a)
#<body #X1007074c (:cube 10.0 20.0 30.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
eusgl$ (read a)
eusgl$
eusgl$
 
 
 
稲葉

~> eusgl
readmacro object packsym common constants stream string loader pprint process hashtab array mathtran eusdebug eusforeign coordinates tty history toplevel trans comp builtins par intersection geoclasses geopack geobody primt compose polygon viewing viewport viewsurface hid shadow bodyrel dda helpsub eushelp xforeign Xdecl Xgraphics Xcolor Xeus Xevent Xpanel Xitem Xtext Xmenu Xscroll Xcanvas Xtop Xapplwin 
connected to Xserver DISPLAY=localhost:0.0
X events are being asynchronously monitored.
pixword RGBHLS convolve piximage pbmfile image_correlation oglforeign gldecl glconst glforeign gluconst gluforeign glxconst glxforeign eglforeign eglfunc glutil gltexture glprim gleus glview �[34m
EusLisp 8.26 for Cygwin  created on Fri Jun 19 09:20:27 2009�[0mtoiv-undefined 
eusgl$ (setq a (piped-fork "eusgl"))
#<io-stream #X10239dfc>
eusgl$ connected to Xserver DISPLAY=localhost:0.0
X events are being asynchronously monitored.
�[34m
EusLisp 8.26 for Cygwin  created on Fri Jun 19 09:20:27 2009�[0m
nil
eusgl$ (format a "(list 'a 'b)~%")
nil
eusgl$ (read a)
readmacro
eusgl$ (read a)
object
eusgl$ (until (eq 'eusgl$ (read a)))
nil
eusgl$ (read a)
(a b)
eusgl$ (format a "(list 1 2)~%")
nil
eusgl$ (read a)
eusgl$
eusgl$ (read a)
(1 2)
eusgl$ (read a)
eusgl$
eusgl$ (format a "(+ 1 2)~%")
nil
eusgl$ (read a)
3
eusgl$ (read a)
eusgl$
eusgl$

 ```

garaemon, 2015-02-12:
> if we use output=log and bootstrap message goes to stdout, then I think everyone is happy.

Yes I think so too but I have seen a lot of programs has "output=screen" and everybody checks output in terminal rather than rqt_console.

garaemon, 2015-02-13:
https://github.com/jsk-ros-pkg/jsk_common/pull/773

If the problem is GUI, this should work...

k-okada, 2015-02-18:
if you're roseus user, https://github.com/jsk-ros-pkg/jsk_roseus/pull/238 is another solution, which if you run roseus with '-q' option, you will not see any bootstrap message, but it also disable your stdin input, so if you invoke roseus with -q
```
$ roseus -q hoge.l 
;; loading roseus("fba5d29") on euslisp((9.10 kokada-t440s Wed Feb 18 20:15:37 JST 2015 213f06f 1.0.2))
(isatty nil)
(isatty nil)
[ INFO] [1424261647.783173583]: cell* ROSEUS_EXIT(context*, int, cell**)
```
it will execute commads in input file, then exit. Do not drop in interpreter.
You can also use this within shebang like `#/usr/bin/env roseus -q`

wkentaro, 2016-04-30:
+1 but I'd like to use quiet mode with stdin.
Is that impossible?

k-okada, 2016-05-06:
please check https://github.com/jsk-ros-pkg/jsk_roseus/pull/238,
btw what do you mean by `quiet mode with stdin.`?
`echo "(print 0") | roseus` ? or just run `roseus` and did not generate any
message, including prompt?