euslisp/EusLisp #120: converstion of interger to string

kochigami, 2015-07-15:
I would like to convert interger to string.
For example,
```
(setq name "fig")
(setq count 1)
(convert 1 to "1"
 count -> converted1) <- I don't know how to do it
(setq picture-name (concatenate string name converted1))
(picture-name)-> "fig1"
```
 I found there is ```write-to-string``` method in common lisp.
Could you tell me if there is some way like this in euslisp?
I'm very sorry if there is some explanation in jmanual, but I could not find it.

snozawa, 2015-07-15:
How about this?
```
$(setq count 1)
$(format t "Fig.~d" count) ;; 
"Fig.1"
$(setq picture-name (format nil "Fig.~d" count))) ;; If nil is specified, return string
"Fig.1"
```

kochigami, 2015-07-15:
I appreciate your help.
It worked very well !!

closed 2015-07-15 by kochigami