euslisp/EusLisp #133: dotimesの中でcontinue文

wkentaro, 2015-09-01:
dotimes blockの中でcontinue文と同じ動作ををさせるにはどうすればよいでしょうか。

wkentaro, 2015-09-01:
solved thanks to @furushchev 
use `(return)` in loop block.

wkentaro, 2015-11-06:
returnだといわゆるbreakをしてしまうので、reopenしました。

furushchev, 2015-11-06:
tagbodyを使うのかな。どういうループでしょうか？例を挙げて欲しいです。
そもそもcontinueというのはほとんど使わないほうが綺麗に書けるはずです。

k-okada, 2015-11-06:
http://lists.gnu.org/archive/html/help-gnu-emacs/2005-12/msg00194.html

コンティニューは無いみたいですね．なにか奥深い理由でもあるのかな．tagでgo すれば良いのかな．
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/

wkentaro, 2015-11-06:
@furushchev 
https://github.com/start-jsk/jsk_apc/pull/959/files
が実際の変更（continueを諦めて大きなif文をつくった）です。
本当はhttps://github.com/start-jsk/jsk_apc/pull/959/files#diff-d3243f6b70b01caa0512314eb37e2a33R47 のあとにcontinueを入れたいというものです。

furushchev, 2015-11-06:
case文の中身を関数にするのは駄目？

furushchev, 2015-11-06:
こういう感じでやれば分かりやすいかも知れません
https://gist.github.com/4d161a3229506764db2d

wkentaro, 2015-11-06:
スマックはフォローしてなかったですが、少し見てみます。

wkentaro, 2015-11-06:
それはそれとして、continueと同じことがdotimesやwhileの中でやれば出来るはずだと思うので、知っていたら教えてください。

wkentaro, 2015-11-06:
jmanualかreadthedocsに書いておきたいです。(出来ないなら出来ないと)

furushchev, 2015-11-06:
eusだといくつか制約があるので、ちょっと汚いですが、例えば以下のような感じになるかと思います。

```lisp
(setq l (list 0 1 2 3 4 5 6 7 8 9))

(defmacro dotimes2 (var cnt &rest body)
  `(let ((,var 0))
     (while (< ,var ,cnt)
            (prog1
                (block :continue
                    ,@body)
                (inc ,var)))))

(dotimes2 i (length l)
          (if (oddp (elt l i))
              (return-from :continue) ;; equivalent to continue
              (print (elt l i))))
```

以下コメント

> jmanualかreadthedocsに書いておきたいです。(出来ないなら出来ないと)

これはlispの一般的な考え方なので、例えば「pythonではmacroがありません。」とpythonのドキュメントに書くのと同じ気がします・・・

wkentaro, 2015-11-06:
一般的な考え方というのは，「continueを使うくらいならrefactorしなさい」というのがですか？

wkentaro, 2015-11-06:
> 例えば「pythonではmacroがありません。」とpythonのドキュメントに書くのと同じ気がします・・・

僕はそれでも書いたほうがいいと思う派ですね．
pythonでいうと，「caseはありませんのでif+elifを使いましょう」というのはdocs.python.orgに書いてあると思います．

「pythonではmacroがありません」と書いていないのはmacroを知っているようなhackerがいちからdocumentを読むことは想定していないからではないかなと．でもcaseやcontinueはCの初級の授業や本でも必ず出るから書いておくと少しは混乱を避けられるだろうという考えです．
(今はjmanualでcontinueと調べても何も出ない．)
[image]

k-okada, 2015-11-07:
http://euslisp-docs.readthedocs.org/en/latest/euslisp/manual/#iteration

あたりにかいたらいいのかな．ちゃんと明示的に意識していないですが，loopかwhileのなかでtagができていたら，そのまま
go出来る気がする
(while t
  (go nanika) ;; exit from loop
  (go nakanaka) ;; continue to next loop
 )
みたいな感じで（もっとマシにして）かけばいいかな．
dotimes, dolist, loop
はぜんぶマクロで中はwhileだあっけ？であああれえば，whileでそのことと，whileのサンプルとしてコードがあればイイ？

wkentaro, 2015-11-07:
こんなかんじでしょうか。
https://gist.github.com/wkentaro/e1143aa452ad7ce9a29f

whileのコンディションのところのマクロでの書き方があっているのかわかりませんが。。
とりあえず動きます。

wkentaro, 2015-11-07:
> while test {form}*
[ 特殊 ]
test が non-NIL と評価されている間、 form は、繰返し評価される。 while は、 form のまわりに NIL と
名付けられるブロックを自動的に確定する特殊書式である。 return は、そのループから抜け出すために
使用することができる。

とありますが、

```
(setq cnt -1)
(while
  (< cnt 10)
  (incf cnt)
  (when (eq (mod cnt 3) 0) (go nil))
  (print (format nil "now at ~a" cnt))
  )
```
とするとエラーになります。

k-okada, 2015-11-09:
> form のまわりに NIL と名付けられるブロックを自動的に確定する特殊書式である

このblockは何のために有るんだろう？

furushchev, 2015-11-09:
>> form のまわりに NIL と名付けられるブロックを自動的に確定する特殊書式である
> このblockは何のために有るんだろう？

`return`でループから抜けるため（Cで言う`break`のため）だと思います。

```lisp
(setq l (list 0 1 2 3 4 5 6 7 8 9))

#|
;; 本当はこうしたいけど、eusにはmacroletがない。。
(defmacro dotimes2 (vars &rest body)
  (let ((endvar (gensym "DOTIMES"))
        (conttag (gensym "DOTIMES")))
    `(let ((,(car vars) 0)
           (,endvar ,(cadr vars)))
       (macrolet ((continue () (return-from ,conttag t)))
         (while (< ,(car vars) ,endvar)
            (prog1
                (block ,conttag
                    ,@body)
                (inc ,(car vars))))))))

(dotimes (i (length l))
  (if (oddp (elt l i))
      (continue)
    (print (elt l i))))
|#

;; 現実的には以下くらいが思いつきそうです。でもdotimes2が2回以上重なると残念。
(defmacro dotimes2 (vars &rest body)
  (let ((endvar (gensym "DOTIMES")))
    `(let ((,(car vars) 0)
           (,endvar ,(cadr vars)))
       (while (< ,(car vars) ,endvar)
            (prog1
                (block :continue
                    ,@body)
                (inc ,(car vars)))))))

(dotimes2 (i 10)
  (if (oddp (elt l i))
      (return-from :continue)
    (print (elt l i))))
```

furushchev, 2015-11-09:
> `;; 本当はこうしたいけど、eusにはmacroletがない。。`

hyperspecをみると、common lispでは`flet`のなかでもblockから抜けられるようですね。
http://clhs.lisp.se/Body/s_block.htm

```lisp
 ;; Contrast behavior of this example with corresponding example of CATCH.
 (block b
   (flet ((b1 () (return-from b 1)))
     (block b (b1) (print 'unreachable))
     2)) =>  1
```

eusにこの機能が入る&&macroを定義すれば、continueがどうしても欲しい人にも使えるかもしれませんね。

commit 22bd03fe (2015-11-21): Describe about jumping to next iteraction in block

Closes #133
closed 2015-11-21 by k-okada