euslisp/EusLisp #138: eusのcopy-object でsegmentation faultする

h-kamada, 2015-09-11:
(ros::roseus "test")
(require :pr2-interface "package://pr2eus/pr2-interface.l")
(load "package://jsk_demo_common/euslisp/attention-observation.l")
(require :pr2-move "package://jsk_demo_common/euslisp/pr2-move.l")
(load "models/room73b2-scene.l")
(load "lisp/comp/builtins.l")
(room73b2)
(dolist (obj (send *room73b2* :objects) nil)
  (ros::ros-info "obj:~A" obj)
  (setq obj (copy-object obj))
  )

を走らせると、copy-object でsegmentation faultする。
gdbすると次のようになる。

;p=pointer?(0x68dd140)

Program received signal SIGSEGV, Segmentation fault.
0x000000000041b262 in copyobj (ctx=0x7510b0, org=0x68e8798) at leo.c:560
560       if (p_marked(org)) return(cpvec[intval(x)]);
(gdb)

garaemon, 2015-09-11:
gdbでwhereと入力しよう

snozawa, 2015-09-11:
エラー出力はすべてはりつけると良いと思います。
あとは、
- 以前は動いてたコードが急にこうなったのか、今はじめて書いてみたコードでこうなったのか。
- もう少し小さい再現コードにできるか。例えば、dolistのobj一個目からcopy-objectに失敗しているのか、何個めかでなるのか。その失敗しているeusモデルだけをloadしてそのモデルだけを作って同じことをしても失敗するか

などがわかると良いですね。

furushchev, 2015-09-11:
@h-kamada とオフラインで一緒にデバッグした結果、`load`していたライブラリのうち
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_mongo/euslisp/mongo-client.l#L70
の`insert`という関数の名前がどこかとコリジョンしていたようです。
(`insert-msg`という名前に変えるPR出しました。 https://github.com/jsk-ros-pkg/jsk_roseus/pull/363 )
同様の経緯で
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_mongo/euslisp/mongo-client.l#L84
の`delete-by-id`も`delete`から変更を余儀なくされているのですが、
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_mongo/euslisp/mongo-client.l#L18-L19
のように`package`を分けていても衝突するとダメなのは仕様なのでしょうか？

garaemon, 2015-09-11:
仕様的には、packageを分けていたらぶつかってもいいけど、userパッケージのようなデフォルトでグローバルな空間になるようなものに関しては、shadowしなくてはいけない、とかあった気がする。

roseus上げただけだとinsertは定義されていないように見える
```
$ apropos 'insert
;; geometry::insert-intersections  Function
;; x:insertmodifiermapentry  Function
;; insert-record  Function
;; insert-record2  Function
;; insert
;; :insert  Constant=:insert
;; :insert-char  Constant=:insert-char
;; :insert-newline  Constant=:insert-newline
;; :insert-string  Constant=:insert-string
;; line-insert  Function
;; list-insert  Function
```

garaemon, 2015-09-11:
loadする順番を変えると、動く気がする
```lisp
(ros::roseus "test")
(require :pr2-interface "package://pr2eus/pr2-interface.l")
(require :pr2-move "package://jsk_demo_common/euslisp/pr2-move.l")
(load "lisp/comp/builtins.l")
(load "models/room73b2-scene.l")
(load "package://jsk_demo_common/euslisp/attention-observation.l")
(room73b2)
(dolist (obj (send *room73b2* :objects) nil)
(ros::ros-info "obj:~A" obj)
(copy-object obj)
)
```

garaemon, 2015-09-11:
きのせいじゃなかった。
この順番だと動いてしまう
```lisp
(ros::roseus "test")
(require :pr2-interface "package://pr2eus/pr2-interface.l")
(require :pr2-move "package://jsk_demo_common/euslisp/pr2-move.l")
(load "lisp/comp/builtins.l")
(load "package://jsk_demo_common/euslisp/attention-observation.l")
(load "models/room73b2-scene.l")
(room73b2)
(dolist (obj (send *room73b2* :objects) nil)
(ros::ros-info "obj:~A" obj)
(copy-object obj)
)
```

garaemon, 2015-09-11:
これは動かない
```
(ros::roseus "test")
(load "models/room73b2-scene.l")
(load "package://jsk_demo_common/euslisp/attention-observation.l")
(room73b2)
(dolist (obj (send *room73b2* :objects) nil)
(ros::ros-info "obj:~A" obj)
(copy-object obj)
)
```

これは動く
```
(ros::roseus "test")
(load "package://jsk_demo_common/euslisp/attention-observation.l")
(load "models/room73b2-scene.l")
(room73b2)
(dolist (obj (send *room73b2* :objects) nil)
(ros::ros-info "obj:~A" obj)
(copy-object obj)
)
```

garaemon, 2015-09-11:
僕のデバッグによると、
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_mongo/euslisp/mongo-client.l#L11
この`defvar`があるかないかで動くか動かないかが変化するんだけどなんでだろう

garaemon, 2015-09-11:
`ros::get-param`のなかでデフォルト値を渡すとleo.c:`copyobj`を呼び出すみたいで、これが行けないのではないか(cpvecを壊している?)

というわけでros::get-paramを繰り返し読んでみると、SEGVするらしい.
https://github.com/jsk-ros-pkg/jsk_roseus/pull/365

COPYOBJ=copy-objectを使うようにすると、解決する
https://github.com/jsk-ros-pkg/jsk_roseus/pull/364

furushchev, 2015-09-11:
https://github.com/jsk-ros-pkg/jsk_roseus/pull/363はcloseしても良さそうですね。

garaemon, 2015-09-11:
ついでにdeleteのことに行っておくと、`delete`は組み込み関数なのでmongoパッケージでshadowしないといけないはず。
roseus.lがヒントになるかもしれませんが、面倒なので別名関数にしちゃう今のアプローチが楽でいいと思います。

garaemon, 2015-09-11:
参考にして欲しいのでデバッグの経緯を書いておくと
* 順番にコードを小さくしていく
  * デバッグの基本。なるべく最小のコードに持っていく。特にeusのような動的な言語だとデバッガが弱いことが多いので大事。
* @h-kamada の言うとおりloadするかしないかで変化する
* しかもloadする順番で影響を受けるのでかなり怪しい
* `insert`関数は名前が被っていないので悪さはしないはず
* 他の部分をコメントアウトすると`(defvar *mongo-database* (ros::get-param "mongodb/database" "test"))`の行があるかないかで変わる
* これを(setq *mongo-database* "test")とすると動く。
* `ros::get-param`が怪しい
* roseus.cppを見ると中でcopyobjを使っている。
  * これはこのバグ報告で落ちている場所と同じ
* `copyobj`と`COPYOBJ`の関係を見ると`copyobj`に行く前に色々と準備しているっぽい
* `copyobj`は循環参照を回避するためにコピー済みのオブジェクトはマークをつけたりするので、結構ナイーブな関数という前提知識がある. (deep copyは一般的に循環を考えると実装が面倒)
* `copyobj`を直接呼ぶとeusの環境を壊してしまいそうなものが`COPYOBJ`の後処理とかにちらほら見える
* `copyobj`が前提としている副作用とか事前の環境の状態はちょっと読み解くのが時間がかかりそうなので`ros::get-param`をCOPYOBJで書き換えてみる. 
* うごいた、よかった

h-kamada, 2015-09-11:
```
(ros::roseus "test")
(require :mongo-client "package://roseus_mongo/euslisp/mongo-client.l")
(load "package://jsk_maps/src/eng2-rooms.l")
(load "models/room73b2-scene.l")
(room73b2)
(setq obj (copy-object (car (send *room73b2* :objects))))
```

読み込んでいるファイルの中身まで調べて行った結果、これで動かないことを確認しました。
その上で、
`package://jsk_maps/src/eng2-rooms.l `
の中で
`(load "models/room73b2-scene.l")`
が呼ばれていることに気づきました。（重複して呼んでしまっていた）
これが悪さをしている可能性もあるでしょうか？

garaemon, 2015-09-11:
2回loadしてもまぁ問題はないはずです。
ですがdefclassを2回やってしまうときに2回目と1回目でslotの順番が違っているとちょっとやばい、とかあった気がします。コンパイルしている場合だけかな?

garaemon, 2015-09-11:
一応jsk-ros-pkg以下に関してはcopyobj問題はもうなさそう。

```
$ gmilk --all copyobj
../../../../../../hydro_parent/src/euslisp/lisp/c/collector.c:9: * FIXED: copyobj in leo.c
../../../../../../hydro_parent/src/euslisp/lisp/c/collector.c:13: *       write-barriers (copyobj() etc.)
../../../../../../hydro_parent/src/euslisp/lisp/c/eus_proto.h:225:extern pointer copyobj(context */*ctx*/, pointer /*org*/);
../../../../../../hydro_parent/src/euslisp/lisp/c/eus_proto.h:227:extern pointer COPYOBJ(context */*ctx*/, int /*n*/, pointer /*argv*/*);
../../../../../../hydro_parent/src/euslisp/lisp/c/external_markbits.c:54:  /* copyobj in leo.c:619 < leo.c:632 */
../../../../../../hydro_parent/src/euslisp/lisp/c/external_markbits.c:63: * copyobj in leo.c:619 < leo.c:632
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:549:pointer copyobj(ctx,org)
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:594:        clone->c.obj.iv[1]=copyobj(ctx,x);
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:595:        if (s>0) clone->c.obj.iv[0]=copyobj(ctx,org->c.obj.iv[0]);
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:596:        for (i=2; i<s; i++) clone->c.obj.iv[i]=copyobj(ctx,org->c.obj.iv[i]);
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:599:        clone->c.vec.v[0]=copyobj(ctx,x);
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:600:        for (i=1; i<s; i++) clone->c.vec.v[i]=copyobj(ctx,org->c.vec.v[i]);
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:603:        clone->c.vec.v[0]=x; /*copyobj(ctx,x) fails */
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:631:pointer COPYOBJ(ctx,n,argv)
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:643:  if ((b=(pointer)eussetjmp(cpyjmp))==0) b=copyobj(ctx,a); 
../../../../../../hydro_parent/src/euslisp/lisp/c/leo.c:737:  defun(ctx,"COPY-OBJECT",mod,COPYOBJ);
../../../../../../hydro_parent/src/euslisp/lisp/comp/builtins.l:69:(def-builtin-entry 'LISP:COPY-OBJECT "COPYOBJ")
../../../../../../hydro_parent/src/euslisp/lisp/comp/builtins.l.c:77:"(def-builtin-entry 'LISP:COPY-OBJECT \"COPYOBJ\") ",
../../../../../../hydro_parent/src/euslisp/lisp/comp/comp.c:8575:       w=(pointer)COPYOBJ(ctx,1,local+3); /*copy-object*/
../../../../../../hydro_parent/src/euslisp/lisp/geo/geobody.c:3733:     w=(pointer)COPYOBJ(ctx,1,local+3); /*copy-object*/
../../../../../../hydro_parent/src/euslisp/lisp/geo/primt.c:1456:       w=(pointer)COPYOBJ(ctx,1,local+7); /*copy-object*/
../../../../../../hydro_parent/src/euslisp/lisp/geo/primt.c:4093:       w=(pointer)COPYOBJ(ctx,1,local+13); /*copy-object*/
../../../../../../hydro_parent/src/euslisp/lisp/geo/primt.c:4571:       w=(pointer)COPYOBJ(ctx,1,local+1); /*copy-object*/
../../../../../../hydro_parent/src/euslisp/lisp/image/piximage.c:424:   w=(pointer)COPYOBJ(ctx,1,local+0); /*copy-object*/
../../../../../../hydro_parent/src/euslisp/lisp/l/string.c:1770:        w=(pointer)COPYOBJ(ctx,1,local+1); /*copy-object*/
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:12592:Objective-C/instm/NSCoder/encodeBycopyObject:
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:15073:C/func/-/FSCopyObjectAsync
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:15074:C/func/-/FSCopyObjectSync
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:15197:C/func/-/FSPathCopyObjectAsync
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:15198:C/func/-/FSPathCopyObjectSync
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:18473:C/func/-/ICACopyObjectData
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:18474:C/tdef/-/ICACopyObjectDataPB
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:18475:C/func/-/ICACopyObjectPropertyDictionary
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:18476:C/tdef/-/ICACopyObjectPropertyDictionaryPB
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:18477:C/func/-/ICACopyObjectThumbnail
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:18478:C/tdef/-/ICACopyObjectThumbnailPB
../../../../../../../gprog/emacs.d/modules/elpa/packages/yasnippet/extras/imported/objc-mode/Support/alldefs.txt:36733:C/func/-/NSCopyObject
../roseus.cpp:1292:      ret = COPYOBJ(ctx,1,argv+1);
../roseus.cpp:1331:      ret = COPYOBJ(ctx,1,argv+1);
../../../../ros_naoqi/naoqi_bridge/naoqi_tools/scripts/blender/io_export_ogreDotScene.py:5456:            def duplicate_object(scene, name, copyobj):
../../../../ros_naoqi/naoqi_bridge/naoqi_tools/scripts/blender/io_export_ogreDotScene.py:5465:                ob_new.data = copyobj.data.copy()
../../../../ros_naoqi/naoqi_bridge/naoqi_tools/scripts/blender/io_export_ogreDotScene.py:5466:                ob_new.location = copyobj.location
../../../../ros_naoqi/naoqi_bridge/naoqi_tools/scripts/blender/io_export_ogreDotScene.py:5467:                ob_new.rotation_euler = copyobj.rotation_euler
../../../../ros_naoqi/naoqi_bridge/naoqi_tools/scripts/blender/io_export_ogreDotScene.py:5468:                ob_new.scale = copyobj.scale
```

h-kamada, 2015-09-11:
解決したということでcloseしますね！

closed 2015-09-11 by h-kamada