euslisp/EusLisp #141: [要望] unittestでassertがfailしたときにcheck関数に何が渡されたか表示したい

furushchev, 2015-10-15:
たとえば

```
  (assert (eq (send (send *sm* :start-state) :name) :a))
```

とすると、

```
TEST-NAME: test-methods
  now testing...
start testing [test-methods]
[ERROR] test (eq (send (send *sm* :start-state) :name) :a) failed ... ( mstart testing [test-methods]
a (&amp;optional args) (ros::ros-info called :d) :success)
over) :fail)
)
.�).
```

というように`test (eq (send (send *sm* :start-state) :name) :a) failed`からfailしたassert箇所はわかりますが、この時に`(send (send *sm* :start-state) :name)`は何だったのかをエラー表示して欲しい。
とくにREPLが使えないときに重宝すると思います。

http://docs.python.jp/2/library/unittest.html
pythonには`assertXXXX`というメソッドがいくつも定義されているが、これらはそういう目的を満たすためだと考えられる。

実装案：

- assertの仕方はいっぱいあるので、`(assert (eq A B))`タイプと`(assert (equal A B)`タイプくらいを特別に実装しておく
  https://github.com/euslisp/EusLisp/pull/140
- assertに渡されるS式の最初はチェック関数で有る可能性が高いので、assertがfailしたときに、チェック関数の引数を評価したものを表示する  
  e.g. `(assert (check A B))` -> `(setq args (mapcar #'eval (cdr pred))) (assert (apply #'check args))`としてassert failのときにargsを表示する
- check関数から呼び出される関数をすべて`trace`してassert failのときに表示する

k-okada, 2015-10-19:
```
(setq ret (send (send *sm* :start-state) :name))
(assert (equal ret :a) (format nil "state name is ~A, which is expected to be :a" ret))
```
ではダメかな．

furushchev, 2015-10-19:
それで大丈夫です。ほぼすべてのassertがそう書いてあるのでもう少し記述量が減らないかと思っての提案でした。

closed 2015-10-19 by furushchev

k-okada, 2015-10-19:
もし他に，必要だと感じている人がいたら，またreopenしてみてください．