euslisp/EusLisp #142: integerのバイナリデータをfloatに変換したい

kyawawa, 2015-10-15:
pythonで言うところの
```python
In [17]: struct.unpack('f', struct.pack('i', 1105199104))
Out[17]: (28.0,)
```
をeuslispで実行したいのですが，どのようにすればよいでしょうか．

YoheiKakiuchi, 2015-10-16:
こんなかんじでしょうか。
~~~
(defun test (intval) (let ((str (make-string 4))) (sys::poke intval str 0 :integer) (sys::peek str 0 :float)))

$ (test 1105199104)
28.0
~~~

kyawawa, 2015-10-16:
ありがとうございます

closed 2015-10-16 by kyawawa

garaemon, 2015-10-16:
このときのintとfloatって、30bit(62bit?)になるんでしたっけ?
それとも一般的なint/floatのように32(64bit)になるんでしたっけ?

◉  ryohei

YoheiKakiuchi, 2015-10-16:
64bitと32bitで違ってくると思う。
コードの中まで確認してないですが、
64bitのeusなら eus上での 整数/浮動小数点数 は long/double なので、
int/floatを使ったら、バイナリにするときにC言語のレベルでキャストされている。
したがって、データとしては32bit使えている。

32bitでint/float, 64bitでlong/doubleを使った場合は、30/62bit分のデータが
書き、読みされる。書くときは、C言語で読めるバイナリになっている。

garaemon, 2015-10-16:
つまり、32bitマシンだとして、
pokeでbignumだけれども32bit以内のものを書き込んだ場合でもきちんと動くという事でしょうか?

YoheiKakiuchi, 2015-10-16:
そうですね。
32/64bit以内のbignumは32/64bitの整数として書けます。
30/62bitより大きい数はbignumとして読み出されます。

garaemon, 2015-10-16:
@kyawawa ドキュメントに付け足しておきましょう

kyawawa, 2015-10-16:
わかりました．
JMANUALのどのあたりでしょうか？

また，かくべき内容は
　処理系が32bit(or64bit)のときに，32bit(or64bit)以内の整数は32(or64bit)の整数として，それ以上はbignumとして読み出される
　32bit(or64bit)以内でもbignumでも，垣内さんのpeek/pokeを使う方法で，整数値から浮動小数点数に戻すことができる
といったものでしょうか．

> 32/64bit以内のbignumは32/64bitの整数として書けます。

bignumとしてかいても，数値が32/64bit以内なら32/64bitの整数になってbignumでなくなるということでしょうか．

また，今回想定してるケースでは
　32bitマシンのロボットで整数値のデータをとる（ただしlong型)
　64bitマシンのeuslisp上でそれを読みだして浮動小数点数にする(double)
としてるので，どのみちbignumはでてこない範囲で読み出せてるという理解でいいでしょうか．

garaemon, 2015-10-16:
> わかりました．
> JMANUALのどのあたりでしょうか？
peek/pokeのあたりでいいんじゃないかな。
「たとえばビット表記の浮動小数点数を読み出す場合はxxxx. この場合、利用できるビット幅の最大値はアーキテクチャに依存し、
32bit OSであれば32bit, 64bit OSであれば64bitである。」

◉  ryohei

kyawawa, 2015-10-16:
ありがとうございます．
書いてみます．