euslisp/EusLisp #147: マクロとlambda

k-okada, 2015-11-02:
```

lマクロとlambda関数について質問させてください。
下に示すようなeusのコードを実行すると、

1.irteusgl$ demo
[ INFO] [1446446798.232458366]: a:goto-front-of-fridge b:(:rarm :larm)
[ INFO] [1446446798.232677039]: a:base-go-to-fridge b:(:base :larm)
nil
2.irteusgl$ func 
(lambda-closure nil 15480480 0 (userdata) (ros::ros-info "a:~A b:~A" (car ag) (cadr ag)))
3.irteusgl$ funcall func 2
;; Segmentation Fault.
;; in (funcall func 2)
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=1797480240 x=6b235e00 addr=ec36a0

となります。
hoge1が実行されている際、a , bにはリスト化した中身がしっかりと渡されていますが、lambda関数の中身に関してはa, bの値はcar ag などのようにしか反映されていないようです。

どのように記述すればfunc の中身の,a ,bにあたるものが反映されるでしょうか。

(defmacro hoge1 (a b)
  `(let ()
     (ros::ros-info "a:~A b:~A" ,a ,b)
     (setq func #'(lambda (userdata)
                    (ros::ros-info "a:~A b:~A" ,a ,b)
                    ))))

(defmacro hoge2 (arg)
  `(let ()
     (dolist (ag ,arg)
       (hoge1
        (car ag) (cadr ag))
       )))

(defun demo ()
   (hoge2
    '((goto-front-of-fridge (:rarm :larm))
      (base-go-to-fridge (:base :larm))
      ))
   )
```

k-okada, 2015-11-02:
hoge2 を defun にしてみるとどうでしょうか？ @hkamada

h-kamada, 2015-11-02:
```
(defmacro hoge1 (a b)
  `(let ()
     (ros::ros-info "a:~A b:~A" ,a ,b)
     (setq func #'(lambda (userdata)
                    (ros::ros-info "a:~A b:~A" ,a ,b)
                    ))))

(defun hoge2 (arg)
  (let ()
    (dolist (ag arg)
      (hoge1
       (car ag) (cadr ag))
      )))

(defun demo ()
  (hoge2
   '((goto-front-of-fridge (:rarm :larm))
     (base-go-to-fridge (:base :larm))
     ))
  )
```

ということでしょうか？
defmacroと同じ結果が得られます。

```
1.irteusgl$ demo
[ INFO] [1446456948.164796699]: a:goto-front-of-fridge b:(:rarm :larm)
[ INFO] [1446456948.164898512]: a:base-go-to-fridge b:(:base :larm)
nil
2.irteusgl$ func
(lambda-closure nil 30201600 0 (userdata) (ros::ros-info "a:~A b:~A" (car ag) (cadr ag)))
```

k-okada, 2015-11-02:
```
(defmacro hoge1 (a b)
  `(let ()
     (ros::ros-info "hoge1 a:~A b:~A" ,a ,b)
     (setq func (list 'lambda '(userdata)
                      (list format t "func a:~A b:~A~%" ,a ,b))
                    )))
```
は？

h-kamada, 2015-11-02:
```
;;raw --> error
(defmacro hoge1 (a b)
  `(let ()
     (ros::ros-info "a:~A b:~A" ,a ,b)
     (setq func #'(lambda (userdata)
                    (ros::ros-info "a:~A b:~A" ,a ,b)
                    ))))
;;answer1
(defmacro hoge1 (a b)
  `(let ()
     (ros::ros-info "hoge1 a:~A b:~A" ,a ,b)
     (setq func (list 'lambda '(userdata)
                      (list 'format t "func a:~A b:~A~%" `',,a `',,b)))
     ))

;;answer2
(defmacro hoge1 (a b)
  `(let ()
     (ros::ros-info "hoge1 a:~A b:~A" ,a ,b)
     (setq func `(lambda (userdata)
                      (format t "func a:~A b:~A~%" ',,a ',,b)))
     ))

(defun hoge2 (arg)
  (let ()
    (dolist (ag arg)
      (setq func (hoge1
       (car ag) (cadr ag)))
      )))

(defun demo ()
  (hoge2
   '((goto-front-of-fridge (:rarm :larm))
     (base-go-to-fridge (:base :larm))
     ))
  )
```

出来ました。
answer1 かanswer 2で、どちらも出力が
```
1.irteusgl$ demo
[ INFO] [1446464705.815808922]: hoge1 a:goto-front-of-fridge b:(:rarm :larm)
[ INFO] [1446464705.816038746]: hoge1 a:base-go-to-fridge b:(:base :larm)
nil
2.irteusgl$ func
(lambda (userdata) (format t "func a:~A b:~A~%" 'base-go-to-fridge '(:base :larm)))
3.irteusgl$ funcall func 2
func a:base-go-to-fridge b:(:base :larm)
nil
```
となります。

ところで
```
4.irteusgl$ #'(lambda () nil)
(lambda-closure nil 0 0 nil nil)
```
となり、出力される関数が`lambda`と`lambda-closure` で異なりますが、この２つは何が異なるのでしょうか。

h-kamada, 2015-11-02:
```
(defmacro hoge1 (a b)
  `(let ()
     (ros::ros-info "hoge1 a:~A b:~A" ',a ',b)
     (setq func '(lambda (userdata)
                      (format t "func a:~A b:~A~%" ',a ',b)))
     ))

(defmacro hoge2 (arg)
  `(let ()
    (dolist (ag ,arg)
      (setq func
            (eval `(hoge1 ,(car ag) ,(cadr ag))))
      )))

(defun demo ()
  (hoge2
   '((goto-front-of-fridge (:rarm :larm))
     (base-go-to-fridge (:base :larm))
     ))
  )
```
```
1.irteusgl$ demo
[ INFO] [1446466256.136489047]: hoge1 a:goto-front-of-fridge b:(:rarm :larm)
[ INFO] [1446466256.136569157]: hoge1 a:base-go-to-fridge b:(:base :larm)
nil
2.irteusgl$ func
(lambda (userdata) (format t "func a:~A b:~A~%" 'base-go-to-fridge '(:base :larm)))
3.irteusgl$ funcall func 2
func a:base-go-to-fridge b:(:base :larm)
nil
```

hoge2の方を変えてこんな感じの書き方もありました。
見た目的にはこっちの方が綺麗な気もしなくもないです。

YoheiKakiuchi, 2015-11-02:
以下にもありますが、現状ではlet内でclosure (lambda式) を使うとうまくいきません。
マクロでも展開されてletが入っていると同じです。
https://github.com/euslisp/EusLisp/issues/71

以下のように、lambda-closure nil xxxx となっていて、 xxxx が0でないとlet内で作られていて、letの外にでるとseg faultします。 let内では使えるはずです。
```
(lambda-closure nil 15480480 0 (userdata) (ros::ros-info "a:~A b:~A" (car ag) (cadr ag)))
```