euslisp/EusLisp #149: classのインスタンスをそのまま保存して後でロードする方法

wkentaro, 2015-11-07:
などありますか。特にrobot-interfaceクラスのインスタンスに対して使いたいです。

やりたいことは、ロボットのステート（認識結果やアングルベクターなど）がslotsに入っているので、
testを行うときにそのままロードして各メソッドのテストをする、ということです。
rosbagを再生してもいいですが、さっとできる方がいいのと、
あるメソッドのための初期姿勢に移動するために準備（必要なメソッドを順に呼ぶ）が必要なのをどうにかしたいというものです。

k-okada, 2015-11-07:
```
おっと，重要な情報なので，書きます．

ソケットごしにインスタンスをやりとりしたい場合には，
*print-object*と，*print-circle*というスペシャル変数を
使えばコンスにいちいち置き換えずに，内部データを表示することが
でき，ソケットでやりとりすることができるようになります．

通常の#<xxxx ... >はずらずら表示されると大変なのでこうなっている
だけです．

*print-object*がtだと，インスタンスの内容を表示するという意味で
*print-circle*がtだと，自己参照しているような再帰構造データの場合に
ラベルをつけて参照する出力を行うようにしてくれます．
たとえば，bodyなどは内部にidenticalなデータを持っているので
*print-circle*をtにせずに，*print-object*だけtにすると
無限に表示を続けてしまいます．

e:/home/inaba/eus/test> eusg
1.eusg$ (make-cube 1 2 3)
#<body #Xa1c832c (:cube 1.0 2.0 3.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
2.eusg$ (setq a (piped-fork "eusg"))
#<io-stream #Xa1c5644>
3.eusg$ (format a "(+ 1 2)~%")
nil
4.eusg$ (read a)
1.eusg$
5.eusg$ (read a)
3
6.eusg$ (format a "(setq *print-object* t *print-circle* t)~%")
nil
7.eusg$ (read a)
2.eusg$
8.eusg$ (read a)
t
9.eusg$ (format a "(make-cube 1 2 3)~%")
nil
10.eusg$ (read a)
3.eusg$
11.eusg$ (read a)
#<body #Xa1c3b44 (:cube 1.0 2.0 3.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>

と手元のcygwin eusではなります．
一番最後が向こう側のeusgで作ったbodyが送られてこちらの
eusgのデータとして読み込むことができたというものです．

で，途中何回もreadしているのは，向こうのeusgのプロンプトを
読み飛ばしているためのものですが，以前のSunのeusでは
piped-forkされたeusは，プロンプトを出さないようになっていて
プロンプトの読み飛ばしを必要としませんでした．
直した方がいいですね．> 岡田君，水内君

後，今，eusgで試していますが，jskrbeusglのように
ライブラリがロードされる際にいろいろメッセージを表示して
いるとして，そのメッセージの中にコメント　の セミコロンをつけて
いないものがあると，同じようにpiped-forkした際に読み飛ばして
やらないといけなくなります．
メッセージは*error-output*に出してあれば*standard-output*を
出ないので読み飛ばしをする必要がなくなります．

稲葉

手元でやると(make-body 1 2 3)はこういう表示になっています．
#1= が再帰データのラベル定義で，内部でこのラベルを参照
した形になっているわけです．

2.jskrbeusgl$ (setq *print-object* t *print-circle* t)
t
3.jskrbeusgl$ (make-cube 1 2 3)
#1=#J(body ((:volume . 5.99999) (:centroid . #f(0.0 5.898382e-09
-2.980234e-08))) #J(array nil #f(1.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0
1.0) 2 nil 0 3 3 nil nil nil nil nil) #f(0.0 0.0 0.0) nil nil
#J(coordinates nil #J(array nil #f(1.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0
1.0) 2 nil 0 3 3 nil nil nil nil nil) #f(0.0 0.0 0.0)) #1# t
#J(bounding-box #f(-0.508 -1.016 -1.524) #f(0.508 1.016 1.524))
(#16=#J(face nil #f(0.0 0.0 1.0) -1.5 t (#24=#J(edge nil #3=#f(-0.5
1.0 1.5) #2=#f(0.5 1.0 1.5) #5=#J(face nil #f(0.0 1.0 0.0) -1.0
t (#21=#J(edge nil #8=#f(0.5 1.0 -1.5) #2# #11=#J(face nil #f(1.0 0.0
0.0) -0.5 t (#18=#J(edge nil #10=#f(0.5 -1.0 -1.5)
#19=#f(0.5 -1.0 1.5) #12=#J(face nil #f(0.0 -1.0 0.0) -1.0 t
(#14=#J(edge nil #6=#f(-0.5 -1.0 -1.5) #15=#f(-0.5 -1.0 1.5)
#4=#J(face
nil #f(-1.0 0.0 0.0) -0.5 t (#23=#J(edge nil #3# #7=#f(-0.5 1.0 -1.5)
#4# #5# 1.5708 0) #13=#J(edge nil #6# #7# #9=#J(face nil
#f(0.0 0.0 -1.0) -1.5 t (#22=#J(edge nil #7# #8# #9# #5# 1.5708 0)
#20=#J(edge nil #8# #10# #9# #11# 1.5708 0) #17=#J(edge nil #10#
#6# #9# #12# 1.5708 0) #13#) (#7# #8# #10# #6# #7#) #f(0.0 0.0 -1.0)
-1.5 nil #1# #9# (:bottom)) #4# 1.5708 0) #14# #25=#J(edge nil
#15# #3# #4# #16# 1.5708 0)) (#3# #7# #6# #15# #3#) #f(-1.0 0.0 0.0)
-0.5 nil #1# #4# (:side 0 0 :-x)) #12# 1.5708 0) #17# #18#
#26=#J(edge nil #19# #15# #12# #16# 1.5708 0)) (#15# #6# #10# #19#
#15#) #f(0.0 -1.0 0.0) -1.0 nil #1# #12# (:side 0 1 :-x)) #11#
1.5708 0) #20# #21# #27=#J(edge nil #2# #19# #11# #16# 1.5708 0))
(#19# #10# #8# #2# #19#) #f(1.0 0.0 0.0) -0.5 nil #1# #11# (:side
0 2 :-x)) #5# 1.5708 0) #22# #23# #24#) (#2# #8# #7# #3# #2#) #f(0.0
1.0 0.0) -1.0 nil #1# #5# (:side 0 3 :-x)) #16# 1.5708 0) #25#
#26# #27#) (#2# #3# #15# #19# #2#) #f(0.0 0.0 1.0) -1.5 nil #1# #16#
(:top)) #9# #5# #11# #12# #4#) (#22# #24# #20# #21# #27# #17#
#18# #26# #23# #13# #14# #25#) (#8# #2# #10# #19# #7# #6# #15# #3#)
(#f(0.5 1.0 -1.5) #f(0.5 1.0 1.5) #f(0.5 -1.0 -1.5) #f(0.5 -1.0
1.5) #f(-0.5 1.0 -1.5) #f(-0.5 -1.0 -1.5) #f(-0.5 -1.0 1.5) #f(-0.5
1.0 1.5)) t nil ((:cube 1.0 2.0 3.0)))
```

k-okada, 2015-11-07:
実際は
http://euslisp-docs.readthedocs.org/en/latest/euslisp/manual/#dump-objects
を使う．ここ，サンプルプログラムをドキュメントに入れてもいいかも

wkentaro, 2015-11-07:
詳細にありがとうございます。
後で*ri*で試してみます。

今まで話に上がらなかったので、jmanualに書いていないものだと思っていました、すみません。

因みにドキュメントの編集はdoc/jlatex以下のものに対して行えばいいでしょうか。
それともroseusのようにdocstringから生成のようなものがありますか。

k-okada, 2015-11-07:
2015-11-07 15:06 GMT+09:00 Kentaro Wada <notifications@github.com>:

> 因みにドキュメントの編集はdoc/jlatex以下のものに対して行えばいいでしょうか。
>

doc/jlatex, doc/latex を変更して下さい．

--
◉ Kei Okada

wkentaro, 2015-11-07:
1. make-cube
2. dump
3. load
できました。
https://gist.github.com/wkentaro/703b05d95354b8d2b8ac

wkentaro, 2015-12-05:
`*ri*`でやろうとすると、
```
$ dump-loadable-structure "STAR-ri-STAR-recognized-bin-objects.l" *ri*
$ load "STAR-ri-STAR-recognized-bin-objects.l"
/home/baxter/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: number expected in read-float-array
```
と言われてロードができないです。

wkentaro, 2015-12-05:
```
9.E2-irteusgl$ btrace
euserror
euserror
euserror
euserror
read-float-array
read-float-array
(apply #'ros::load-org-for-ros ros::fullname args)
(apply #'ros::load-org-for-ros ros::fullname args)
(let ((ros::fullname fname)) (when (substringp "package://" fname) (setq ros::fullname (ros::resolve-ros-path fname)) (when ros::*compile-message* (let* ((ros::urlname (url-pathname fname)) (package-name (send ros::urlname :host)) (ros::path-name (format nil "~A_~A" package-name (substitute 95 47 (concatenate string (subseq (send ros::urlname :directory-string) 1) (send ros::urlname :name))))) (ros::old-module (find ros::path-name *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless ros::old-module (let* ((ros::ppath (unix:getenv "CMAKE_PREFIX_PATH")) (dir (format nil "~A/share/roseus/ros/~A" (subseq ros::ppath 0 (position 58 ros::ppath)) package-name))) (unless (probe-file dir) (unix:mkdir dir)) (compiler:compile-file-if-src-newer ros::fullname (format nil "~A/~A" dir ros::path-name)) (return-from load (load (format nil "~A/~A.so" dir ros::path-name) :entry (format nil "___~A" ros::path-name)))))))) (if (null ros::fullname) (error "file ~s not found" fname)) (apply #'ros::load-org-for-ros ros::fullname args))
t
10.E2-irteusgl$ load "lisp/l/array.l"
t
11.E2-irteusgl$ btrace
euserror
euserror
euserror
euserror
read-float-array
read-float-array
(apply #'ros::load-org-for-ros ros::fullname args)
(apply #'ros::load-org-for-ros ros::fullname args)
(let ((ros::fullname fname)) (when (substringp "package://" fname) (setq ros::fullname (ros::resolve-ros-path fname)) (when ros::*compile-message* (let* ((ros::urlname (url-pathname fname)) (package-name (send ros::urlname :host)) (ros::path-name (format nil "~A_~A" package-name (substitute 95 47 (concatenate string (subseq (send ros::urlname :directory-string) 1) (send ros::urlname :name))))) (ros::old-module (find ros::path-name *loaded-modules* :key #'lisp::load-module-file-name :test #'equal))) (unless ros::old-module (let* ((ros::ppath (unix:getenv "CMAKE_PREFIX_PATH")) (dir (format nil "~A/share/roseus/ros/~A" (subseq ros::ppath 0 (position 58 ros::ppath)) package-name))) (unless (probe-file dir) (unix:mkdir dir)) (compiler:compile-file-if-src-newer ros::fullname (format nil "~A/~A" dir ros::path-name)) (return-from load (load (format nil "~A/~A.so" dir ros::path-name) :entry (format nil "___~A" ros::path-name)))))))) (if (null ros::fullname) (error "file ~s not found" fname)) (apply #'ros::load-org-for-ros ros::fullname args))
t
12.E2-irteusgl$ load "STAR-ri-STAR-recognized-bin-objects.l"
/home/baxter/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: number expected in (apply 'float-vector list)
13.E3-irteusgl$ ,list
(-nan -nan -nan)
1
```

wkentaro, 2015-12-05:
maybe `-nan` is not numbe

k-okada, 2015-12-05:
Nanはどのスロットに入っているかな

wkentaro, 2015-12-05:
normal, model-normalなどで#f(-nan -nan -nan)となっています．
[image]

k-okada, 2015-12-09:
これどうしましょうか．
#f(nan inf nan)
みたいなのもロードできるようにするのは一つの解にはなると思うけど．

wkentaro, 2015-12-09:
@furushchevによると、`*nan*`はloadできるということなので、print-objectのときにそうなるようにするという手もあるかと思います。

YoheiKakiuchi, 2015-12-09:
現状では、 ```#f(*nan* )```もloadできていない（数字が期待されている）です。

wkentaro, 2015-12-13:
なるほど，`#f(nan)`は`(float-vector *nan*)`に直す必要があるということですか．

ちなみに`nan`が入るのは普通でしょうか．他のロボットでdumpやloadがうまくいくなら，単純にbaxter-interfaceのバグということになりますね．

YoheiKakiuchi, 2015-12-14:
nanが入っているのはfaceのようなので、モデルを作るときのfaceの面積が小さかったり、normalが計算できなかったりしているんだと思います。
そちらで、nanが入ったときに、normalならnan nan nanでなくて、0 0 0にするなどの方法はあるかと思います。

PR #150 (2016-03-20): Add sample for dumping object functions
Closes #149

いまいち`dump-object`と`dump-structure`の違いがわからないので適切か見てください。
commit 9dff361c (2016-03-20): Add sample for dumping object functions

Closes #149
closed 2016-03-20 by k-okada