euslisp/EusLisp #163: "Outer circuit not found"  というエラーがでるのですが、これはどういうことでしょうか？

HarukiCho, 2015-12-25:
```
1.eusgl$ (defun make-door nil
  (let (b1)
    (setq b1 (make-cube 5 40 90))
    (setq c1 (make-cylinder 5 5))
    (setq c2 (make-cylinder 5 5))
    (send c1 :rotate (deg2rad 90) :y)
    (send c1 :locate #f(5 15 0))
    (send c2 :rotate (deg2rad 90) :y)
    (send c2 :locate #f(-5 15 0))
    (setq b1 (body+ b1 c1))
    (setq b1 (body+ b1 c2))
    (send b1 :locate #f(5 20 0))
    (send b1 :set-color :blue)
    b1))

make-door
2.eusgl$ nil
2.eusgl$ make-door
eusgl 0 error:  outer circuit not found
 in (body+ b1 c1)
```

HarukiCho, 2015-12-25:
https://github.com/euslisp/EusLisp/blob/74fff34afa261db392800f1f2ea7518ee4cadfff/lisp/geo/compose.l#L299
でしょうか

k-okada, 2015-12-25:
おそらく面と面が接している状態でbody+しているんだとおもいます．
ちょっと食い込ませてbody+するしか無い気がします．

HarukiCho, 2015-12-25:
あ

HarukiCho, 2015-12-25:
ありがとうございます

Naoki-Hiraoka, 2020-06-29:
面と面が接していない状態でbody+しても，Outer circuit not foundエラーが出ることがありました

```
(defun test (x)
  (setq *a* (make-cube 100 100 100))
  (setq *b* (make-cube (- 100 x) (- 100 x) (- 100 x)))
  (send *b* :translate (float-vector 0 0 x))
  (setq *c* (body+ *a* *b*))
  (objects (list *c*))
  )

(test 2) ;;NO ERROR
(test 1) ;;ERROR ( outer circuit not found)
```

`(setq *epsilon* 1e-5) ` `(setq *contact-threshold* 8e-5)`などと，\*epsilon\*や\*contact-threshold\*の値を小さくすると解決しました．