euslisp/EusLisp #167: GSoC 2016: Return Barrier Garbage Collector

kunaltyagi, 2016-03-20:
Hi, I'm interested in implementing the RGC.

I have read the [paper](http://www.yuasa.kuis.kyoto-u.ac.jp/~yuasa/ilc2002/paper.pdf) and the relevant [Makefile](https://github.com/euslisp/EusLisp/blob/master/lisp/Makefile.rgc).
@k-okada What should I do next?

PS: [README.rgc](https://github.com/euslisp/EusLisp/blob/master/lisp/README.rgc) is unreadable.

k-okada, 2016-03-23:
What's next? write proposal, and if you have time, please try to compile
with rgc option.
sorry for late, I have fixed the README.rgc, but that is in Japanese, do
you think Google Translate work for you?

kunaltyagi, 2016-03-25:
I have compiled with rgc option, and have submitted a draft proposal.
Google Translate wasn't perfect, but it worked good enough for me to read it.