euslisp/EusLisp #183: 複数のIKを並列に解く

chiwunau, 2016-06-15:
例えば1000個の独立した位置に対するIKを解く場合、並列に解くにはどうすればいいでしょうか
Euslispにはmulti-thread対応していると思うんですが、サンプルなどはないでしょうか

eusを何個か立ち上げて解を書き出すということもできると思うが、もっといい方法があればご教示ください。

k-okada, 2016-06-16:
遅くなりました．
マルチプロセスとマルチスレッドのサンプルを以下におきました．
マルチスレッドにしても，並列に走っている状態だと，それぞれのスレッドのロボットモデルは別のものを使わないといけなくなるので，
１０００個並列にしたければ１０００こロボットモデルを作ることになるかと思います．今だとroseusでサーバ作ってそこに投げるのが良いのではないでしょうか．
https://github.com/jsk-ros-pkg/jsk_control/tree/master/jsk_ik_server
というのもあるので確認してみてください．

https://gist.github.com/k-okada/510a12509427e5af818697a8572fe2ac#file-multi-process-example-l

https://gist.github.com/k-okada/510a12509427e5af818697a8572fe2ac#file-multi-thread-example-l

snozawa, 2016-06-16:
（内部ですが）
https://github.com/jsk-ros-pkg/jsk-ros-pkg-unreleased/issues/52
と関連しますが、マップをつくるためのプログラム
https://github.com/jsk-ros-pkg/euslib/tree/master/demo/ik/motion-map
と、その中で複数processのEuslispを作って、mapcar的に使えるようにしていたものがありました（うえだくんさく）
https://github.com/jsk-ros-pkg/euslib/blob/master/demo/ueda/parallel/parallel-lib.l

chiwunau, 2016-06-21:
Thank you, I will study about them.

chiwunau, 2016-12-13:
@k-okada 先生が教えて頂いた
https://gist.github.com/k-okada/510a12509427e5af818697a8572fe2ac#file-multi-thread-example-l
の multi-thread-example.l の方を倣って並列処理を書いていますが、Bugが出ています。

こちらのコードで各スレッドにそれぞれPR2とドアのモデルを与えて色々な計算をしています。
https://gist.github.com/chiwunau/00c6545def7f3f5f2c2a9d724b8be83b

コードではスレッド３つを作っていて、
1スレッド当たりに短い計算(IK数個）をさせるとと問題なく終わりますが、
各スレッドに長い計算(IK数十から百個くらい）をさせると、
途中に３つのスレッドのうちの一つか二つが下のエラーが出て死んでしまう問題が発生しています。
```
/home/leus/ros/hydro_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 8 error:  go tag not found in (go #:do*55474)
```
なにが原因なのか心当たりがあるのでしょうか

k-okada, 2016-12-14:
> ac#file-multi-thread-example-l

(update-loaded-nodes :robot robot :door door :nodes nodes)
  (robot-pose-sampling :robot robot :door door :nodes nodes))
をもっと簡単にして参照セットでエラーがるものを作ってみてくれると嬉しいです．

chiwunau, 2016-12-19:
頑張ってデバックしてみたら解決したようです。原因は自分のミスで
irtmodel.lとirtdyna.lの中にあるmethodsをそのまま手元のコードにコピーしていて、
シングルスレッドの時には問題はなかったが、マルチスレッドの時に二重参照?の問題で上のエラーが出ます。

なので手元のmethodsを削除したらうまく行きました。

k-okada, 2016-12-19:
それはよかったです．ただ、今後も同じようにトラブルに見舞われる人がいると困るので、
作っていることを
https://github.com/jsk-ros-pkg/jsk-ros-pkg-unreleased
のdemo/au というディレクトリをつくって、そこに入れて下さい．