euslisp/EusLisp #203: substringp関数に対するsubstring関数があると嬉しい時があるかもしれない

h-kamada, 2016-10-21:
substringpの実装は一致したらreturnするようになっているが、文字列の中に何個一致するものがあるか調べたい事があるかもしれず、そういった際に使えるsubstring関数があるとありがたいかもしれない。
https://github.com/euslisp/EusLisp/blob/9af32e9f989108e0a77de140a891daf09216e7dd/lisp/l/string.l#L125-L131
今回は結局使わなかったので報告のみ。

h-kamada, 2016-10-21:
string-left-trimは一見使えそうだが、最初の文字が該当しないと返ってきてしまうので使えない。
````
1.irteusgl$ substringp "hoge" "ahogefugahogea"
t
2.irteusgl$ string-left-trim "hoge" "ahogefugahogea"
"ahogefugahogea"
````

snozawa, 2016-10-21:
irteusには文字列関係の関数が少ないので、あると良いと思ったことは個人的にも多いです。
内部コードになりますが、euslib/jsk/cregexp.cに関数があるので、
```
roseus
(jsk)
```
すると
```
irteusgl$ (1- (length (reg-split "hoge" (format nil " ~A " "hogefugahoge"))))
2
```
のようになりますが、これをcregexp.cをirteusにうつしてくるのか、
また別なことをかんがえるかを検討する必要がありそうです

furushchev, 2016-10-21:
@h-kamada 
> 文字列の中に何個一致するものがあるか調べたい事があるかもしれず

これは例えばどういう問題のときに必要になるでしょうか？

k-okada, 2016-10-25:
> f09216e7dd/lisp/l/string.l#L125-L131

練習問題として実際に実装してみましょう．

snozawa, 2016-10-25:
ちなみに
https://github.com/euslisp/EusLisp/issues/39
は関係ありますか？
pythonにあり、かつjskrbeusglなどで使えてる機能をirteusかEuslispにもってきたいときには、
議論をしつつ、

> 練習問題として実際に実装してみましょう．

OKがでれば実装してみる、という方針で良いでしょうか。

k-okada, 2016-10-26:
思いついたものは実装し、実装したものはPRを送る、というのが方針でしょうか．

ただ、jskeusglにあるものをirteusに持ってきてほしいという声は幾つも上がりますが、実際にPRを送っても少し時間がたつと議論が収束したり、必要ありません、となるので、
結局、それほど必要なかったのでは？と思ったりします．

https://github.com/euslisp/jskeus/pull/248
https://github.com/euslisp/jskeus/pull/234
https://github.com/euslisp/jskeus/pull/225
https://github.com/euslisp/jskeus/pull/261

snozawa, 2016-10-26:
> 結局、それほど必要なかったのでは？と思ったりします．

確かにこれはありますね。
あげていただいたものを見ると、PRを出してくれてた人が卒業しちゃう、パターンもありますね。
個人的に多いのが、機能としては必須で普段から使ってるのですが、
- `(jsk)`をよぶとつかえる
- `euslib/jsk`以下のファイルをそのまま単体でloadすると使える（こっちのほうはほぼ副作用ない場合がおおい。`(jsk)`を全部よんでしまうのと比較して）

という機能の場合、irteusやeuslispに昇格されてなくても使えてしまうので、
https://github.com/euslisp/jskeus/pull/261
のように進んでないのかもしれないと思いました。

k-okada, 2016-10-26:
それはそうだけど、それを許すと結局
jskirteus使いましょう -> なんか汚いし見たく無いです -> irteus2 を作りました！
と成長がないので、もう少し広く見て、これは波及効果が高いからこっちね、これはまだ尖った部分だからこっちね、をやるのが重要だと思います

snozawa, 2016-10-26:
なるほど、わかりました。

h-kamada, 2016-10-27:
@furushchev 返信が遅れました。
例えばですが
````
(defun substring (sub str)
  (let ((cnt 0))
    (do ((i (- (length str) (length sub)))
         (l (length sub))
         (j 0 (1+ j)))
        ((> j i) cnt)
      (when (string-equal sub str :start2 j  :end2 (+ j l))
        (incf cnt)
        ))))
(setq a (list
         "xxhogehogexx"
         "xxhogefugaxx"
         "xxfugahogexx"
         "xxfugafugaxx"
         ))
(dolist (aa a)
  (when (eq (substring "hoge" aa) 2)
    (ros::ros-info "~A" aa)
    ;; (do something)
    ))
````
として実行すると
````
> [ INFO] [1477546542.160349674]: xxhogehogexx
````
といったような形で使うようなことを考えてました。

h-kamada, 2016-10-27:
実際に使おうと思った場面としては、dotファイルの記述の中で、
````
(actionA robot1 ....) -> (actionB robot1 .....)
(actionC robot1 ....) -> (actionD robot2 .....)
(actionE robot2 ....) -> (actionF robot1 .....)
(actionG robot2 ....) -> (actionH robot2 .....)
````
のような記述がなされているものを書き換えて、
````
(actionC robot1 ....) -> (actionD robot2 .....)
(actionE robot2 ....) -> (actionF robot1 .....)
````
とすることを考えていました。
その際のやり方として、`robot1`もしくは`robot2`の記述が一文に２つ出てくるものは除く、といったことをやろうとしていました。

closed 2016-11-11 by h-kamada