euslisp/EusLisp #210: 関数ブロックからのreturn-fromが効かない時がある

furushchev, 2016-11-01:
```lisp
(defun func-a (&rest args)
  (when args
     (return-from func-a t))
  nil)
```

という関数を定義したとして、実行すると、

```lisp
(func-a) :: => nil
(func-a 1) ;; => t
```

が期待されると思いますが、

```lisp
(func-a 1)
roseus-error: no such block in (return-from func-a t), exitting...
```

というエラーになってしまうことがたまにある気がします。
再現できる最小セットを見つけていますが、現時点ではわからないです。

mmurooka, 2016-11-01:
遭遇したことはないです．

ちなみに，roseus を上げて，関数の定義を貼り付けて
```
(dotimes (i 1000000) (print (list i (func-a t))))
```
しましたが，何事もなく100万回呼べてしまいました．

furushchev, 2016-11-01:
見つけました。

```lisp
;; test-func.l
 (defun func-a (&rest args)
   (warning-message 1 "called func-a with args: ~A~%" args)
   (when args
     (warning-message 1 "entered when block~%")
     (return-from func-a t))
   (warning-message 1 "not entered when block~%"))

 (defun call-func (func-name)
   (let (sym func)
     (setq sym (read-from-string func-name))
     (setq f `(lambda-closure nil 0 0 (x)
                              (apply ',(car sym) ',(cdr sym))))
     (funcall f nil)))
```

```bash
$ (load "test-func.l")
t
$ (call-func "(func-a)")
called func-a with args: nil                                                                                         
not entered when block
nil
$ (call-func "(func-a 1)")
called func-a with args: (1)                                                                                         
entered when block
irteusgl: ERROR th=0 no such block  in (return-from func-a t)
```

mmurooka, 2016-11-01:
再現しました．

furushchev, 2016-11-01:
具体的にどこで困るのかというと、`task_compiler`を使って、pddlからstate-machineが自動生成されて、各ノードに関数がプロパティとしてbind・実行されるのですが、関数のbind部分がこのようになっていました。

https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/roseus_smach/src/pddl2smach.l#L60

コミットを辿って行くと、最初からこのスタイルだったようです。
https://github.com/jsk-ros-pkg/jsk_roseus/commit/629dd55486d6a0e71c5e7c6fb3192718b94f2025#diff-5fe9984d196b26f7af90f462687547bfL54
https://github.com/jsk-ros-pkg/jsk_roseus/commit/88c4f6076689223dd937dec1cd70a5bcb0f0cba1

- @YoheiKakiuchi 
authorはmanabuさんですが、関連コミッタとしてどういう経緯でこの`lamnda-closure 0 0`を使うスタイルになっているのかご存知ですか？
- @YoheiKakiuchi @k-okada この書き方をした時にも一応`apply`で関数をコールしていますが、関数ブロックがないのは正しい挙動でしょうか？
- 何か回避方法をご存知でしょうか？

furushchev, 2016-11-01:
@mmurooka ありがとうございます。何か理由があってこうなったと思うので、もう少し調べてみます。

snozawa, 2016-11-01:
無名関数になってるのが問題な気がしますが，対処療法なのか良い線いってるのか
微妙な解としては，明示的にblockを切ってあげると良いと思います．

```
 (defun func-a (&rest args)
   (block :func-a
          (warning-message 1 "called func-a with args: ~A~%" args)
          (when args
            (warning-message 1 "entered when block~%")
            (return-from :func-a t))
          (warning-message 1 "not entered when block~%")))
```

furushchev, 2016-11-01:
@snozawa ありがとうございます。
確かに実行できました。
が、僕の理解が正しければ,smachから呼ばれる関数すべてにブロックを定義する必要が出てきますでしょうか。
これをコミットをするのは少々気が引けてしまいますが、他に解がなければ致し方ないのでしょうか。。。

snozawa, 2016-11-01:
ああ，ごめんなさい，いっこ関数のいれこを読み飛ばしてた．
だとしたらこうですかね．func-a関数は変更せずにcall-funcを以下にしてためしてください．
```
(defun call-func (func-name)
   (let (sym func)
     (setq sym (read-from-string func-name))
     (setq f `(lambda-closure nil 0 0 (x)
                              (apply #',(car sym) ',(cdr sym)))) ;; 'を#'に
     (funcall f nil)))
```

furushchev, 2016-11-01:
@snozawa おお、できました。ありがとうございます。今までのコードも動くか試してみます。
よく考えたら`,(car sym)`は関数名前空間のシンボルなので、 `#`をつけなくても`apply`で呼べてしまうことが問題ということになるのでしょうか。

YoheiKakiuchi, 2016-11-01:
http://clhs.lisp.se/Body/f_apply.htm
どうかなとおもったけど、applyでシンボルを受け取るのは挙動としては正しそうだ。

> @YoheiKakiuchi authorはmanabuさんですが、関連コミッタとしてどういう経緯でこのlamnda-closure 0 0を使うスタイルになっているのかご存知ですか？

let内でlamda式をつくろうとすると、セグフォする https://github.com/euslisp/EusLisp/issues/71
(Eusでは正しいクロージャーが実装されていない）
これを回避しているはず。

snozawa, 2016-11-01:
> よく考えたら,(car sym)は関数名前空間のシンボルなので、 #をつけなくてもapplyで呼べてしまうことが問題ということになるのでしょうか。

問題なのかな？
少しこんがらがってきましたが，むしろ今回のエラーがでること自体が問題なのかもしれない．
```
$ sbcl
* (defun func-a (&rest args)
   (when args
     (return-from func-a t)))
FUNC-A
* (list
(funcall 'func-a nil)
(funcall 'func-a 1)
(funcall #'func-a nil)
(funcall #'func-a 1)
(apply 'func-a nil)
(apply 'func-a '(1))
(apply #'func-a nil)
(apply #'func-a '(1)))
(T T T T NIL T NIL T)
```
```
$ clisp
[1]> (defun func-a (&rest args)
   (when args
     (return-from func-a t)))
FUNC-A
[2]> (list
(funcall 'func-a nil)
(funcall 'func-a 1)
(funcall #'func-a nil)
(funcall #'func-a 1)
(apply 'func-a nil)
(apply 'func-a '(1))
(apply #'func-a nil)
(apply #'func-a '(1)))
(T T T T NIL T NIL T)
```
```
$ eus0
1.eus0$ (defun func-a (&rest args)
   (when args
     (return-from func-a t)))
func-a
2.eus0$ (list
(funcall 'func-a nil)
(funcall 'func-a 1)
(funcall #'func-a nil)
(funcall #'func-a 1)
(apply 'func-a nil)
(apply 'func-a '(1))
(apply #'func-a nil)
(apply #'func-a '(1)))
eus0 0 error: no such block in (return-from func-a t)
```
をみると，エラーが起こるのはeusだけのよう．

YoheiKakiuchi, 2016-11-01:
> をみると，エラーが起こるのはeusだけのよう．

というより、applyだけだね。funcallは通る。
~~~
(funcall 'func-a 1) ==> t
(apply 'func-a '(1)) ==> error
~~~

シンボルだった時の扱いがfuncallとapplyで違う。
ここに深い意味があるだろうか？

apply
https://github.com/euslisp/EusLisp/blob/master/lisp/c/specials.c#L96-L98
funcall
https://github.com/euslisp/EusLisp/blob/master/lisp/c/specials.c#L128-L129

furushchev, 2016-11-01:
short:
`funcall`を使うと上手く行くので、C言語の実装を見て`funcall`と同じ挙動にしました。
(`funcall`と`apply`は第２引数以降の扱いのみが異なるという前提)

long:

`funcall`の実装では例えfunctionが実体のsymbol pointerが渡されても、function pointerにせずに、実際に関数をコールする`ufuncall`に渡しているのに対し、`apply`ではなぜかfunction pointerにしてから渡していました。
もともと`ufuncall`の中にfunctionが実体のpointerをどう扱うかが（かなり激しいスパゲティコードによって)定義されているので`apply`でも`funcall`と同様にポインタをそのまま渡して見るようにしたところ、解決したように思います。

PR #211 (2016-11-29): [lisp/c/specials.c] apply: delegate refer func pointer to ufuncall
in `APPLY` function, function pointer of symbol pointer is referred (line 97), and the pointer is passed to `ufuncall` which also refers function pointer.
In this pull request, `APPLY` function does NOT refer function pointer and delegates it to `ufuncall`.

This should resolve #210
closed 2016-11-29 by k-okada