euslisp/EusLisp #212: euslispでテキストファイルを開く方法について

kochigami, 2016-11-05:
euslispを使って
ファイルを開き、一行ずつ読み込みたいです。

jmanualを読んだのですが、どうすればいいか分かりませんでした。
サンプルコード等ご存知の方はご教示ください。

このレベルでつまづいています。すみません。
```
with-open-file "abc.txt"
=> /opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: list expected in (with-open-file "abc.txt")

```

kochigami, 2016-11-05:
すみません、自己解決しました。

ファイルを作り同じディレクトリで実行
```
(with-open-file (s "abc.txt" :direction :input)
(let (buff)
(while (setq buff (read-line s nil)) (print buff))))
=> "a b c d"
nil
```

closed 2016-11-05 by kochigami

k-okada, 2016-11-07:
テキストファイルにLisp文法で書いておく、例えば、/tmp/hoge.l を
```
"hoge" "fuga" "aga"
("hoge" "fuga" "aga")
(hoge fuga aga)
```
とすると、
```
(with-open-file (s "/tmp/hoge.l" :direction :input)(while (setq b (read s nil))(print b)))
"hoge"
"fuga"
"aga"
("hoge" "fuga" "aga")
(hoge fuga aga)
```
という風になります．

k-okada, 2022-04-28:
@kochigami @a-ichikura @miyabitane Eusでファイルをどう読むのかな？と思って調べたらこのページが引っかかりました．
さっき僕が考えたファイルを読んで/speech で喋るプログラムの例をつくってみました．文章が長くなるとsleepの部分の調整が必要かもしれません．
https://gist.github.com/k-okada/c0ef7a3a2f4c0984cb014f1f25bad227