euslisp/EusLisp #213: スペースの入った文字列を要素に分けたい

kochigami, 2016-11-05:
```
setq a "neko nao pepper"
"neko nao pepper"
```
となったものをスペースで分けて、
```("neko" "nao" "pepper") ```
としたいのですが、何か良い方法はありますか。

スペースを発見するところは
```
 (dotimes (x (length a))
        (if (eq (elt i x) 32) (print "OK")) ;; 32 == space     
```
のようにして出来ました。

kochigami, 2016-11-05:
文字列をリストにする（数字列になる）
スペースのある場所を数える
スペースの場所に応じ数字列を切り出す
それぞれの数字列を文字列に戻す

という愚直なやり方で解決しました。
もしこの類の関数が本当になくて、必要そうであればドキュメントに書きます。

closed 2016-11-05 by kochigami

furushchev, 2016-11-05:
今更で、すごく適当ですが、

```lisp
(mapcar #'string-downcase (read-from-string (format nil "(~A)" "neko nao pepper")))
```

でもできます。（全部小文字になります。）
確か似たような質問をどこかでみて、そこに回答がいくつかあったのですが、思い出せませんでした。

kochigami, 2016-11-06:
教えて下さりありがとうございました。

- - - 
あまり関係ないですがメモ
```read-from-string ```　と　``` princ-to-string```について
```
(read-from-string "neko")
=> neko ;; stringからs表現（文字のこと？）を取り出す
(princ-to-string 20161106)
=> "20161106"　;; stringになる
```