euslisp/EusLisp #220: eval-when does not work

furushchev, 2016-12-08:
```lisp
;;; test-eval-when.l
(eval-when (:load-toplevel)
  (format t ":load-toplevel~%"))

(eval-when (:compile-toplevel)
  (format t ":compile-toplevel~%"))

(eval-when (:execute)
  (format t ":execute~%"))

(format t "normal toplevel form")
```

```bash
$ eus test-eval-when.l
t
```

k-okada, 2016-12-12:
```
(print "--")
(eval-when (load)
  (format t ":load-toplevel~%"))

(eval-when (compile)
  (format t ":compile-toplevel~%"))

(eval-when (execue)
  (format t ":execute~%"))

(eval-when (eval)
  (format t "eval~%"))
```
use load, compile, eval
```
1.eusgl$ load "/tmp/hoga.l"
"--"
eval
normal toplevel form
t
2.eusgl$ compile-file "hoga.l"
compiling file: hoga.l
"--"
:compile-toplevel
normal toplevel form
"--"
:compile-toplevel
normal toplevel form
gcc -g -c -Dx86_64 -DLinux -Wimplicit -falign-functions=8 -DGCC3  -DGCC  -DTHREADED -DPTHREAD -fpic  -I/opt/ros/indigo/share/euslisp/jskeus/eus/include -O2 hoga.c; ld -shared -build-id -o hoga.so hoga.o
nil
3.eusgl$ (load "hoga.so")
"--"
:load-toplevel
normal toplevel form
#<load-module #X61cb310 "./hoga.so">
4.eusgl$ (load "hoga.l")
"--"
eval
normal toplevel form
t
```

furushchev, 2016-12-12:
@k-okada It works! I'll add to docs soon.

closed 2016-12-12 by furushchev