euslisp/EusLisp #224: dump-structureは同じファイルに対して複数回行えますか

kochigami, 2016-12-15:
``` dump-structure "test.l" 座標 A```
``` dump-structure "test.l" 座標 B```
``` dump-structure "test.l" 座標 C```
を複数回行うと上書きされてしまい、ファイルの中で最後のCしか残らないのですが、
何行もファイルに書いておいて、読み込むときに座標が出てくるような方法はありますか。
一つずつファイルを作るのが正しいのでしょうか。

やりたいこと：
たくさんの座標をファイルに記録して、次にプログラムを立ち上げる時に再利用したいです。
ファイルを読み込んだら勝手に座標系になっていたら楽だと思いました。

今使っている方法：
座標から:pos, :rotの結果を取ってきてファイルに書く
一つずつ読み込んで
``` (setq coords (instance coordinates :init :pos pos :rot rot))```

他に教わった座標系の作り方
```
(setq coords (make-coords :pos #f(1 2 3) :rpy (list 0.1 0.2 0.3)))
#<coordinates #Xbaa49c0  -309.485 145.005 348.903 / -3.012 -0.566 2.707>が手に入った時、-3.012 -0.566 2.707をプログラム上でどのメソッドで得るのかまだ分かっておらず、使えませんでした。
```

（```
(setq b #s(coordinates plist nil rot #2f((-0.224845 0.902382 -0.36763) (-0.350175 -0.426918 -0.833738) (-0.909297 -0.058727 0.411982)) pos #f( 1 2 3)))
```
というものもあるようだ）

YoheiKakiuchi, 2016-12-15:
リストになっているなら、
~~~
(dump-structure "filename" list-of-coordinates)
~~~
とするとリストで保存される。

同じくリストになっているなら、以下のようにすると1行づつ書かれる
~~~
(apply #'dump-structure "filename" list-of-coordinates)
~~~

リストになっていなくて、逐次書き出したいときは、
~~~
(with-open-file (strm "filename" :direction :output)
   ;;書きたい繰り返し処理等。。
   (dump-structure strm obj) ;; 書き出しするとき
   (dump-structure strm obj) ;; 書き出しするとき
   )
~~~

リードするときは、こんな感じ
~~~
(with-open-file (f "filename")
   (while (print (read strm nil nil))
     )
   )
~~~

変数名を含めて保存したいとき、ロードするだけで変数を使えるようにしたいときは以下のようにすると良い
~~~
(setq *coordslist* (list (make-coords) (make-coords) (make-coords)))
(dump-loadable-structure "/tmp/tmp-lodable.l" *coordslist*)
;; テストコード
(makunbound '*coordslist*)
(print *coordslist*) ;; エラーになる
;; このとき *coordslist* シンボルはなにも指していない

;; ロードするとき
(load "/tmp/tmp-loadable.l")
(print *coordslist*)
~~~

Affonso-Gui, 2018-07-25:
Was the problem solved? Can we close this issue?
@k-okada