euslisp/EusLisp #231: OSX で eus64-check が使えるようにする

k-okada, 2017-01-08:
以下を動くようにして確認できるようにしたいです．おそらくdefun-c-callable がうまく動いておらず、holeのあるobjectをひょうじでき無いのではないかと思います．

@wkentaro @furushchev 
```
+++make -C eus/contrib/eus64-check/^M
gcc -m32 -O2 -g -falign-functions=4 -Di386 -Di486 -DLinux -fpic -c test_foreign.c^M
clang: warning: optimization flag '-falign-functions=4' is not supported^M
ESC[1mtest_foreign.c:169:34: ESC[0mESC[0;1;35mwarning: ESC[0mESC[1mformat specifies type 'unsigned long' but the argument has type 'long (*)()' [-Wformat]ESC[0m^M
  printf("set_ifunc, g = %lX\n", g);^M
ESC[0;1;32m                         ~~~     ^^M
ESC[0mESC[1mtest_foreign.c:170:1: ESC[0mESC[0;1;35mwarning: ESC[0mESC[1mcontrol reaches end of non-void function [-Wreturn-type]ESC[0m^M
}^M
ESC[0;1;32m^^M
ESC[0mESC[1mtest_foreign.c:180:35: ESC[0mESC[0;1;35mwarning: ESC[0mESC[1mformat specifies type 'unsigned long' but the argument has type 'double (*)(long, long, long, long, long, long, double, double, double, double, double, double, double, double, double, double, long, long)' [-Wformat]ESC[0m^M
  printf("set_ffunc, gf = %lX\n", gf);^M
ESC[0;1;32m                          ~~~     ^~^M
ESC[0mESC[1mtest_foreign.c:181:1: ESC[0mESC[0;1;35mwarning: ESC[0mESC[1mcontrol reaches end of non-void function [-Wreturn-type]ESC[0m^M
}^M
ESC[0;1;32m^^M
ESC[0mESC[1mtest_foreign.c:184:35: ESC[0mESC[0;1;35mwarning: ESC[0mESC[1mformat specifies type 'unsigned long' but the argument has type 'long (*)()' [-Wformat]ESC[0m^M
  printf("call_ifunc, g = %lX\n", g);^M
ESC[0;1;32m                          ~~~     ^^M
ESC[0mESC[1mtest_foreign.c:189:36: ESC[0mESC[0;1;35mwarning: ESC[0mESC[1mformat specifies type 'unsigned long' but the argument has type 'double (*)(long, long, long, long, long, long, double, double, double, double, double, double, double, double, double, double, long, long)' [-Wformat]ESC[0m^M
  printf("call_ffunc, gf = %lX\n", gf);^M
ESC[0;1;32m                           ~~~     ^~^M
ESC[0m6 warnings generated.^M
gcc -m32 -shared -fpic -falign-functions=4 -o test_foreign.so test_foreign.o^M
irteusgl eus64-test.l^M
ESC[31mXserver connection failedESC[0mBINLOAD cannot dlopen: dlopen(./test_foreign.so, 1): no suitable image found.  Did find:^M
        ./test_foreign.so: mach-o, but wrong architecture^M
Call Stack (max depth: 20):^M
  0: at (load-foreign "test_foreign.so")^M
  1: at (setq *testmod* (load-foreign "test_foreign.so"))^M
```

furushchev, 2017-01-09:
@k-okada 

`./test_foreign.so: mach-o, but wrong architecture`

This is because test_foreign.so is compiled as 32bit linux architecture. (Note that `ARCHDIR` is `Darwin` not `Linux64`) 

However even if compiled as 64bit darwin architecture, calling `defun-c-callable` fails with `Bus Error`(this is equivalent to `Segmentation Fault` in Linux system).

One possible cause was compile option `-falign-functions=<num>` which is available on `GCC`  is not available in `clang`.
 (displayed above: `clang: warning: optimization flag '-falign-functions=4' is not supported`)
I changed it to `-malign-functions=<num>` which seems to be supported on `Clang` (please see below diff. At least, compiler does not complain about it when specified), still it occurs `Bus Error`.

(Note that I also disabled optimization in this diff)

```diff
diff --git a/contrib/eus64-check/Makefile b/contrib/eus64-check/Makefile
index 3ec68ba..d84106f 100644
--- a/contrib/eus64-check/Makefile
+++ b/contrib/eus64-check/Makefile
@@ -4,6 +4,10 @@ all : test
 MARCH=$(shell uname -m)
 
 test_foreign.so : test_foreign.c
+ifeq ($(ARCHDIR), Darwin)
+	gcc -g -malign-functions=8 -Dx86_64 -DDarwin -DLinux -fPIC -c $<
+	gcc -shared -fPIC -malign-functions=8 -dynamiclib -flat_namespace -undefined suppress -o $@ test_foreign.o
+else
 ifeq ($(ARCHDIR), Linux64)
 ##
 	gcc -O2 -g -falign-functions=8 -Dx86_64 -DLinux -fPIC -c $<
@@ -25,10 +29,10 @@ else
 	gcc -m32 -shared -fpic -falign-functions=4 -o $@ test_foreign.o
 endif
 endif
+endif
 
 test: test_foreign.so
 	irteusgl eus64-test.l
 
 clean :
 	\rm -f *.o *.so
-
diff --git a/contrib/eus64-check/eus64-test.l b/contrib/eus64-check/eus64-test.l
index 9cf1560..b32f396 100644
--- a/contrib/eus64-check/eus64-test.l
+++ b/contrib/eus64-check/eus64-test.l
@@ -200,7 +200,7 @@ test-testd = 1.23456
 
   (assert (= (ret-long 123 645000) (+ 123 645000)))
   )
-#|
+
 ;; ret-int
 ;; ret-short
 ;; ret-char
@@ -243,7 +243,7 @@ test-testd = 1.23456
   206 207
   return 0.12345~%")
 (format t "call-ffunc = ~A~%" (call-ffunc))
-|#
+
 
 (run-all-tests)
 (exit)
diff --git a/lisp/Makefile.Darwin b/lisp/Makefile.Darwin
index df7814f..aa56778 100644
--- a/lisp/Makefile.Darwin
+++ b/lisp/Makefile.Darwin
@@ -36,7 +36,7 @@ ifeq ($(GCC_VERSION), 2)
  ALIGN_FUNCTIONS="-malign-functions=4"
 else
  CPU_OPTIMIZE=#-mcpu=i486
- ALIGN_FUNCTIONS=-falign-functions=4
+ ALIGN_FUNCTIONS=-malign-functions=4
  GCC3=-DGCC3 -DGCC4
 endif
 
@@ -46,7 +46,7 @@ ifeq ($(OS_VERSION), 10.5)
  MACHINE=i386
 else
  MACHINE=x86_64
- ALIGN_FUNCTIONS=-falign-functions=8 -fPIC
+ ALIGN_FUNCTIONS=-malign-functions=8 -fPIC
 endif
 DEBUG= -g
 
diff --git a/lisp/comp/comp.l b/lisp/comp/comp.l
index c029408..2e9d498 100644
--- a/lisp/comp/comp.l
+++ b/lisp/comp/comp.l
@@ -27,7 +27,7 @@
 (defparameter *cflags* "")
 (defparameter *defun-list* nil)
 (defparameter *verbose* nil)
-(defparameter *optimize* 2)
+(defparameter *optimize* 0)
 (defparameter *safety* 1)
 (defparameter *space* 0)
 (defparameter *do-cc* t)
@@ -1220,7 +1220,7 @@
  (:compile-file (file &key
 		 (cc-option "")
 		 (optimize *optimize*)
-		 (c-optimize 2)
+		 (c-optimize 0)
 		 (safety *safety*)
 		 (cc *do-cc*)	;nil skips c-compilation
 		 (pic *pic*)	;t forces position independent code 
@@ -1310,8 +1310,8 @@
 		   	 ((memq :sanyo *features*)  " -Dsanyo")
 		   	 ((memq :darwin *features*)
 			  (if (memq :x86_64 *features*)
-			      " -DDarwin -Dx86_64 -DLinux -w -falign-functions=8 "
-			    " -DDarwin -Di386 -DLinux -w -falign-functions=4 "))
+			      " -DDarwin -Dx86_64 -DLinux -w -malign-functions=8 "
+			    " -DDarwin -Di386 -DLinux -w -malign-functions=4 "))
 		   	 ((and (memq :linux *features*)  (memq :gcc3 *features*))
 			  (cond
                            ((memq :x86_64 *features*)
```

furushchev, 2017-01-09:
@k-okada

> おそらくdefun-c-callable がうまく動いておらず、holeのあるobjectをひょうじでき無いのではないかと思います

このテストは通りませんが、holeのあるobjectの表示はできているように思います。

```lisp
irteusgl$ (setq c1 (make-cube 100 100 100))
#<body #X7fdcf90162b0 (:cube 100.0 100.0 100.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
irteusgl$ (setq c2 (make-cube 10 10 1000))
nil
irteusgl$ #<body #X7fdcf90167d8 (:cube 10.0 10.0 1000.0) 0.0 0.0 0.0 / 0.0 0.0 0.0>
irteusgl$ (setq b (body- c1 c2))
#<body #X7fdcf902b970 (:complex -) 0.0 0.0 0.0 / 0.0 0.0 0.0>
irteusgl$ (objects (list b))
;; (make-irtviewer) executed
(#<body #X7fdcf902b970 (:complex -) 0.0 0.0 0.0 / 0.0 0.0 0.0>)
```

[image]

確かに以前は`body-`を使ったオブジェクトを表示できなかった記憶はありますが、どこかの段階で修正されていた気がします。
(もしかしたら https://github.com/euslisp/EusLisp/pull/178 だった気もしますが、違うかもしれません)

furushchev, 2017-01-09:
(MEMO: @k-okada 忘れないようにアサインしていただけると嬉しいです。)

furushchev, 2017-01-09:
現在の疑問
- segfaultはfunction alignmentが原因なのか？
- malign-functionsはclangで効いているのか？(function alignmentが適切に設定されているかどうがをチェックできる最小のプログラムがあるといいのですが・・・)
- なぜ関数アラインメントを8にする必要があるのか(Linux 64bitやosxでは標準では16のようです)

furushchev, 2017-01-11:
もうひとつ考えられるのは`Linux64`では有効になっているリンカオプション`fno-stack-protector -z execstack`がclangだと指定しても無効なオプションになるのも原因かもしれません。
thanks to @YoheiKakiuchi

k-okada, 2017-01-11:
irtviewer が動いているから、ほぼすべてのdefforegin 機能は動いていて、doubleの扱いか、defun-c-callable
か、とにかくなにかひとつだけが動いていない気がします．

eus64-test.l は動いている？これの結果を見たら、どこがおかしいか一発でわかるとおもうんだけど．．．

furushchev, 2017-01-11:
@k-okada 先頭のコメントに貼ったとおり、

```bash
$ make
irteusgl eus64-test.l
configuring by "/Users/furushchev/Development/jskeus-orig/eus/lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin
connected to Xserver DISPLAY=/private/tmp/com.apple.launchd.pY68WQKTDa/org.macosforge.xquartz:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb
;; extending gcstack 0x1098c8000[16374] --> 0x109dbe000[32748] top=3d1a
irtgl irtglc irtviewer
EusLisp 9.22(a89bd07 5d6b9ca1) for Darwin created on feuerzangenbowle.local(Mon Jan 9 09:50:16 JST 2017)

callback function test(integer)
  callback function is set
set_ifunc, g = 7F87468944E0
  expected result: LISP-INTFUNC is called, return 1234
call_ifunc, g = 7F87468944E0
;; Bus Error.
make: *** [test] Error 10
```

となっていて、
https://github.com/euslisp/EusLisp/blob/master/contrib/eus64-check/eus64-test.l#L215
の`set-ifunc`までは動きますが、その後、
https://github.com/euslisp/EusLisp/blob/master/contrib/eus64-check/eus64-test.l#L217
の`call-ifunc`でsegfaultしているようです。

furushchev, 2017-01-11:
https://gist.github.com/furushchev/68130fd3f809810075f6ba7a0d944b2e

k-okada, 2017-01-11:
あ、見逃していました．コンパイルされたコードをnmして関数のポインタを確認するのかな．８バイト境界はeusがデータ型の区別に２bit使うからですね．でも、http://euslisp.github.io/EusLisp/manual-node4.html
でも、https://github.com/euslisp/EusLisp/blob/master/lisp/c/eval.c#L1009 あたりで、必要条件ではなくなっている．pod-addressとかを確認するのかな．．．