euslisp/EusLisp #240: -supplied-p is not supported on euslisp

k-okada, 2017-04-18:
```
* (defun foo (a &optional (b 10 c)) (print (list a b c)))

FOO
* (foo 10)

(10 10 NIL) 
(10 10 NIL)
* (foo 10 20)

(10 20 T) 
(10 20 T)
```
```
18.irteusgl$ (defun foo (a &optional (b 10 c)) (print (list a b c)))
foo
19.irteusgl$ (foo 10)
Call Stack (max depth: 20):
  0: at (list a b c)
  1: at (print (list a b c))
  2: at (foo 10)
  3: at #<compiled-code #X50200c8>
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: unbound variable c in (list a b c)
```
from @Affonso-Gui

k-okada, 2017-04-18:
c.f. https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/robot-interface.l#L903

Affonso-Gui, 2019-04-25:
Supported in cl-compatible branch
https://github.com/euslisp/EusLisp/commit/3e0fa934613cd48bcb16870c5c25910c2b1646d0

closed 2019-04-25 by Affonso-Gui