euslisp/EusLisp #241: return within dolist did not return

k-okada, 2017-04-18:
```
* (dolist (var (list 1 2 3 4 5)) (if (eq var 3) (return var)))

3
```
```
 (dolist (var (list 1 2 3 4)) (if (eq var 3) (return var)))
nil
```
from @Affonso-Gui

k-okada, 2017-04-18:
```
$ (dolist (var (list 1 2 3 4) var) (if (eq var 3) (return var)))
3

$ (block :dolist (dolist (var (list 1 2 3 4)) (if (eq var 3) (return-from :dolist var))))
3
```

Affonso-Gui, 2017-12-01:
Could be solved by using `do` instead of `while` on [dolist definition](https://github.com/euslisp/EusLisp/blob/master/lisp/l/common.l#L156-L166). 

```
	(do ()
	    ((null ,lists) ,(caddr vars))
```
However, this would lead to inefficiency, and therefore is not recommended.
```
Before:
bench (dotimes (i 10000) (dolist (a '(1 2 3 4 5))))
;; time -> 0.279851[s]

After:
bench (dotimes (i 10000) (dolist (a '(1 2 3 4 5))))
;; time -> 0.380254[s]
```

Alternatively, we could check if `while` returned with non-nil values and return that in this case, which would do the job without major changes on efficiency.
```
       (or
        (while ... )
	,(caddr vars))
```
```
(dolist (var (list 1 2 3 4 5)) (if (eq var 3) (return var)))
;; -> 3
```
```
bench (dotimes (i 10000) (dolist (a '(1 2 3 4 5))))
;; time -> 0.293503[s]
```
However, this wouldn't work for cases like `(return nil)`, and it's not very classy.

furushchev, 2018-07-25:
In `sbcl`, `dolist` is expanded as follows:

```lisp
* (pprint (macroexpand '(dolist (i '(1 2 3)) (if (eq i 2) (return t)))))

(BLOCK NIL
  (LET ((#:N-LIST466 '(1 2 3)))
    (TAGBODY
     #:START467
      (UNLESS (ENDP #:N-LIST466)
	(LET ((I (TRULY-THE (MEMBER 3 2 1) (CAR #:N-LIST466))))
          (SETQ #:N-LIST466 (CDR #:N-LIST466))
          (TAGBODY
            (IF (EQ I 2)
		(RETURN T))))
	(GO #:START467))))
```

The original problem is that looping block is not returned as the return value of `dolist`.
Since we have both `tagbody` and `go` in `euslisp`, I think it is good chance to migrate to this implementation if there is no difference in performance.

furushchev, 2018-07-25:
I wrote a simple script for comparison:

```lisp
#!/usr/bin/env roseus
;; dolist-test.l                                                                                                   
;; Author: furushchev <furushchev@jsk.imi.i.u-tokyo.ac.jp>                                                         

(defun iota (n)
  (let ((l (make-sequence cons n)))
    (dotimes (i n)
      (setf (elt l i) i))
    l))

(pprint (macroexpand '(dolist (i (iota 10000)))))
(bench (dolist (i (iota 10000))))
(dolist (i (iota 5)) (print i))

(defmacro dolist (vars &rest forms)
  (let ((lists (gensym "DOLIST"))
        (loop-tag (gensym "DOLIST"))
        (maybe-decl (car forms)))
    (if (and (consp maybe-decl) (eq (car maybe-decl) 'declare))
        (setq forms (cdr forms))
        (setq maybe-decl nil))
    `(block nil
       (let ((,(car vars) nil)
             (,lists ,(cadr vars)))
         ,maybe-decl
         (tagbody ,loop-tag
            (unless (endp ,lists)
              (setq ,(car vars) (pop ,lists))
              ,@forms
              (go ,loop-tag)))))))

(pprint (macroexpand '(dolist (i (iota 10000)))))
(bench (dolist (i (iota 10000))))
(dolist (i (iota 5)) (print i))
```

```lisp
   ((i nil) (#:dolist353 (iota 10000)))
   nil
   (while #:dolist353 (setq i (pop #:dolist353)))
   nil)
;; time -> 0.165225[s]
0
1
2
3
4
(block
   nil
   (let
      ((i nil) (#:dolist10367 (iota 10000)))
      nil
      (tagbody
         #:dolist10368
         (unless
            (endp #:dolist10367)
            (setq i (pop #:dolist10367))
            (go #:dolist10368)))))
;; time -> 0.14457[s]
0
1
2
3
4
```

furushchev, 2018-07-25:
```lisp
#!/usr/bin/env roseus
;; loop-test.l
;; Author: furushchev <furushchev@jsk.imi.i.u-tokyo.ac.jp>

(defun iota (n)
  (let ((l (make-sequence cons n)))
    (dotimes (i n)
      (setf (elt l i) i))
    l))

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;;; DOLIST
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

(warning-message 1 "DOLIST WHILE~%")
(print (dolist (i (iota 5) 'retval) (print i)))
(pprint (macroexpand '(dolist (i (iota 10000)))))
(bench (dolist (i (iota 10000))))

(defmacro dolist (vars &rest forms)
  (let ((lists (gensym "DOLIST"))
        (loop-tag (gensym "DOLIST"))
        (maybe-decl (car forms)))
    (if (and (consp maybe-decl) (eq (car maybe-decl) 'declare))
        (setq forms (cdr forms))
        (setq maybe-decl nil))
    `(block nil
       (let ((,(car vars) nil)
             (,lists ,(cadr vars)))
         ,maybe-decl
         (tagbody ,loop-tag
            (unless (endp ,lists)
              (setq ,(car vars) (pop ,lists))
              ,@forms
              (go ,loop-tag)))
         ,(caddr vars)
         ))))

(warning-message 1 "DOLIST TAGBODY~%")
(print (dolist (i (iota 5) 'retval) (print i)))
(pprint (macroexpand '(dolist (i (iota 10000)))))
(bench (dolist (i (iota 10000))))

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;;; DOTIMES
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;

(setq nrepeat 1000000)
(warning-message 1 "DOTIMES WHILE~%")
(dotimes (i 5) (print i))
(pprint (macroexpand '(dotimes (i nrepeat))))
(bench (dotimes (i nrepeat)))

(defmacro dotimes (vars &rest forms)
  (let ((endvar (gensym "DOTIMES"))
        (loop-tag (gensym "DOTIMES")))
    `(block nil
       (let ((,(car vars) 0)
             (,endvar ,(cadr vars)))
         (declare (integer ,(car vars) ,endvar))
         (tagbody ,loop-tag
            (unless (<= ,endvar ,(car vars))
              ,@forms
              (setq ,(car vars) (1+ ,(car vars)))
              (go ,loop-tag)))
         ,(caddr vars)
         ))))

(warning-message 1 "DOTIMES TAGBODY~%")
(dotimes (i 5) (print i))
(pprint (macroexpand '(dotimes (i nrepeat))))
(bench (dotimes (i nrepeat)))

;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;; $ roseus loop-test.l
;; DOLIST WHILE
;; 0
;; 1
;; 2
;; 3
;; 4
;; retval
;; (let
;;    ((i nil) (#:dolist360 (iota 10000)))
;;    nil
;;    (while #:dolist360 (setq i (pop #:dolist360)))
;;    nil)
;; ;; time -> 0.163418[s]
;; DOLIST TAGBODY
;; 0
;; 1
;; 2
;; 3
;; 4
;; retval
;; (block
;;    nil
;;    (let
;;       ((i nil) (#:dolist10375 (iota 10000)))
;;       nil
;;       (tagbody
;;          #:dolist10376
;;          (unless
;;             (endp #:dolist10375)
;;             (setq i (pop #:dolist10375))
;;             (go #:dolist10376)))
;;       nil))
;; ;; time -> 0.142077[s]
;; DOTIMES WHILE
;; 0
;; 1
;; 2
;; 3
;; 4
;; (let
;;    ((i 0) (#:dotimes20385 nrepeat))
;;    (declare (integer i #:dotimes20385))
;;    (while (< i #:dotimes20385) (setq i (1+ i)))
;;    nil)
;; ;; time -> 0.072827[s]
;; DOTIMES TAGBODY
;; 0
;; 1
;; 2
;; 3
;; 4
;; (block
;;    nil
;;    (let
;;       ((i 0) (#:dotimes20393 nrepeat))
;;       (declare (integer i #:dotimes20393))
;;       (tagbody
;;          #:dotimes20394
;;          (unless (<= #:dotimes20393 i) (setq i (1+ i)) (go #:dotimes20394)))
;;       nil))
;; ;; time -> 1.62785[s]
```

Hmm, it is strange, `dotimes` becomes 20 times slower than `while` version...