euslisp/EusLisp #244: (random 3) is not random

k-okada, 2017-07-27:
by @708yamaguchi

k-okada, 2017-07-27:
1. use `make-random-state t` and `random`
```
$ cat /tmp/hoge.l (setq a (make-random-state t))
(setq ret nil)
(dotimes (i 10)
  (push (random 3 a) ret))
(setq dt (unix:localtime))
(format t "~A    ->~A~%"  (unix:asctime (unix:localtime)) ret)
(exit)
$ while [ 1 ]; do eus2 /tmp/hoge.l  2> /dev/null; sleep 0.3; done
Thu Jul 27 09:38:27 2017
    ->(1 1 0 0 2 0 0 1 1 0)
Thu Jul 27 09:38:27 2017
    ->(1 1 0 0 2 0 0 1 1 0)
Thu Jul 27 09:38:28 2017
    ->(1 0 2 1 2 2 1 0 0 1)
Thu Jul 27 09:38:28 2017
    ->(1 0 2 1 2 2 1 0 0 1)
Thu Jul 27 09:38:28 2017
    ->(1 0 2 1 2 2 1 0 0 1)
Thu Jul 27 09:38:29 2017
    ->(1 1 2 0 0 2 2 2 2 0)
Thu Jul 27 09:38:29 2017
    ->(1 1 2 0 0 2 2 2 2 0)
Thu Jul 27 09:38:29 2017
    ->(1 1 2 0 0 2 2 2 2 0)
Thu Jul 27 09:38:30 2017
    ->(2 0 1 2 2 0 0 0 2 0)
```

k-okada, 2017-07-27:
```
(make-random-state t)
(setq ret (list 0 0 0))
(dotimes (i 10)
  (incf (elt ret (random 3)))
  (print ret))
(print (apply #'+ ret))

(setq ret (list 0 0 0))
(dotimes (i 10)
  (incf (elt ret (mod i 3)))
  (print ret))
(print (apply #'+ ret))

(setq reta (list 0 0 0))
(dotimes (i 10)
  (incf (elt reta (random 3)))
  (print reta))
(print (apply #'+ reta))

(setq reta (list 0 0 0))
(dotimes (i 10)
  (setq j (random 3))
  (incf (elt reta j))
  (print reta))
(print (apply #'+ reta))
(exit 1)
```
outputs 
```
EusLisp 9.23(7bf5d46) for Linux64 created on p40-yoga(Sun Jun 25 17:53:09 JST 2017)
(1 0 0)
(1 0 2)
(1 0 3)
(1 0 1)
(1 0 1)
(1 0 1)
(1 0 1)
(1 0 1)
(2 0 1)
(2 0 2)
4
(1 0 0)
(1 1 0)
(1 1 1)
(2 1 1)
(2 2 1)
(2 2 2)
(3 2 2)
(3 3 2)
(3 3 3)
(4 3 3)
10
(1 0 0)
(1 0 1)
(1 1 1)
(2 1 1)
(2 2 1)
(2 3 1)
(2 3 1)
(2 3 4)
(2 3 3)
(2 4 3)
9
(0 0 1)
(0 1 1)
(1 1 1)
(2 1 1)
(3 1 1)
(3 2 1)
(4 2 1)
(4 3 1)
(4 4 1)
(4 5 1)
10
```
why  `(print (apply #'+ reta))` is not 10?

k-okada, 2017-07-27:
Cc: @YoheiKakiuchi

YoheiKakiuchi, 2017-07-27:
INCF is macro
https://github.com/euslisp/EusLisp/blob/master/lisp/l/common.l#L117-L119

```(incf (elt reta (random 3)))``` may expand to ```(setf (elt reta (random 3)) (1+ (elt reta (random 3))))```

So, ```(incf (elt reta (print (random 3))))``` prints number twice.

Then, by EusLisp
~~~
(setq v (list 0 0 2))
(setq i 0)
(incf (elt v (incf i))) -> 3
(pprint i) -> 2
(pprint v) -> (0 3 2)
~~~

By SBCL, it is correct.
~~~
(setq v (list 0 0 2))
(setq i 0)
(incf (elt v (incf i))) -> 1
(pprint i) -> 1
(pprint v) -> (0 1 2)
~~~

k-okada, 2017-07-27:
I see, thank you

closed 2017-07-27 by k-okada

k-okada, 2017-07-27:
FYI @knorth55
```
EusLisp 9.23(7bf5d46) for Linux64 created on p40-yoga(Sun Jun 25 17:53:09 JST 2017)
((8 10 12) (8 10 12))
((101 100 99) (10 10 9))
((1015 1051 934) (10 10 9))
((9886 10037 10077) (9 10 10))
```
```
$ cat random.l;eus0 random.l 
(make-random-state t)
(dolist (j (list 1 10 100 1000))
  (setq ret (list 0 0 0))
  (dotimes (i (* j 10 (length ret)))
    (setq k (random 3))
    (incf (elt ret k)))
  (print (list ret (mapcar #'(lambda (x) (/ x j)) ret)))
  )
```