euslisp/EusLisp #249: function "string-trim" behaves suspiciously

M-Onitsuka, 2017-10-17:
This is result execution result when I used function "string-trim" on roseus, and I think it's behavior differs from manual.
Is this correct behavior?
```
1.irteusgl$ setq str "test.dat"
"test.dat"
2.irteusgl$ string-trim ".dat" str
"es"
3.irteusgl$ string-trim ".dat" "state.dat"
"state"
4.irteusgl$ string-right-trim ".dat" str
"tes"
5.irteusgl$ string-right-trim ".dat" "state.dat"
"state"
```

Affonso-Gui, 2017-10-17:
Do not worry, there is nothing wrong with euslisp here.

The function `string-trim` removes ANY of the characters present on the first parameter from both the start and the end of the second parameter string. It's more meant to be used like: 
```
(string-trim '(#\Space #\Tab #\Newline) " garbanzo beans
        ") =>  "garbanzo beans"
```
In the "test.dat" example it goes from the beginning like:
1st letter: `t` . Is present in ".dat" so delete it
2nd letter: `e` . Is not present in ".dat" so let it be
3rd letter: `s` . Is not present in ".dat" so let it be
4th letter: `t`: Is present in ".dat" so delete it
and so on, making it return `"es"` in the end. Same for `string-right-trim` and `string-left-trim`.

Please note that `(string-trim "ta.d" "test.dat")` also returns `"es"`.

If you just want to take the file extension you can for instance
```
(setq str "test.dat")
(subseq str 0 (position #\. str))
=> "test"
```
If it must be ".dat" I advice you to have a look at `search` function (not defined in euslisp though).

Some common lisp related links:
http://clhs.lisp.se/Body/f_stg_tr.htm
http://cl-cookbook.sourceforge.net/strings.html

M-Onitsuka, 2017-10-17:
I got it. And thank you for your advice.

closed 2017-10-17 by M-Onitsuka