euslisp/EusLisp #253: Multiple arguments in v+ and v-

Affonso-Gui, 2017-11-21:
The following did not give expected result:
```
(v+ #f(1) #f(10) #f(100))   > #f(11.0)

(v+ #f(1) #f(10) #f(100) #f(1000))  > mismatch argument error 
```
Definitions in `matrix.c` show that `v+` and `v-` 's 3rd argument is used exclusively to store the result. 
In order to sum several vectors, `v++` is defined instead. `v--` does not appear to be defined, though.

For me this seemed to be very counter intuitive and rarely used syntax, but I'm afraid changing it can lead to crashes in upper layer programs.

k-okada, 2017-11-21:
instantiate vector takes cpu time, compare to the calculation,

for example

9.irteusgl$ (let ((a #f(1)) (b #f(10)) (c #f(0))  (m (instance mtimer
:init))) (dotimes (i 10000000)  (v+ a b)) (send m :stop))
2.95625
10.irteusgl$ (let ((a #f(1)) (b #f(10)) (c #f(0))  (m (instance mtimer
:init))) (dotimes (i 10000000)  (v+ a b c)) (send m :stop))
1.7004

please note that

8.irteusgl$ (let ((a #f(1)) (b #f(10)) (c #f(0))  (m (instance mtimer
:init))) (dotimes (i 10000000)  (+ 1 2)) (send m :stop))
1.42103
11.irteusgl$ (let ((a #f(1)) (b #f(10)) (c #f(0))  (m (instance mtimer
:init))) (dotimes (i 10000000)  ) (send m :stop))
0.970572

as a baseline.

I did not know `v++`, and we may add `v--`

Affonso-Gui, 2017-11-21:
I made a brief search and it seems that such a syntax is used in OpenGL's C-coded part, where indeed optimization is needed. However, it doesn't seem to be used at all at lisp part.

Being that the case, wouldn't it be better to define lisp's `v+` as `VPLUSPLUS` and give `VPLUS` another name, for example `v+to`, or leave it undefined? 

This could be easily done by editing [matrix.c](https://github.com/euslisp/EusLisp/blob/master/lisp/c/matrix.c#L1359-L1360), although would affect codes using v++ or v+ with three arguments. 

Tests I've done:
- Defining `VPLUS` as `VPLUSPLUS` led to [outer circuit not found](https://github.com/euslisp/EusLisp/blob/master/lisp/geo/compose.l#L312) error in `make-camera-from-param` function, when starting `pr2`.
- Defining `v+` as `VPLUSPLUS` and leaving `v++` undefined led to no errors, and started pr2 successfully.

YoheiKakiuchi, 2017-11-21:
> Being that the case, wouldn't it be better to define lisp's v+ as VPLUSPLUS and give VPLUS another name, for example v+to, or leave it undefined?

:-1:
This may confuse old people.

There are many use case in lisp/l/coordinates.l lisp/geo/geopack.l

Affonso-Gui, 2017-11-27:
@YoheiKakiuchi Thanks for the feedback. It would indeed be confusing to change functions now, and third argument optimization seems to be used on lisp part as well.

@k-okada I don't think `v--` needs to be added, as it will probably almost never be used (as `v++`).

Closing.

closed 2017-11-27 by Affonso-Gui