euslisp/EusLisp #257: `reverse (cons 1 2) ` returns `(1)`

knorth55, 2018-02-08:
```lisp
32.E3-irteusgl$ (reverse (cons 1 2))
(1)
```
SBCL returns type error for argument.
```lisp
$ sbcl
This is SBCL 1.1.14.debian, an implementation of ANSI Common Lisp.
More information about SBCL is available at <http://www.sbcl.org/>.

SBCL is free software, provided as is, with absolutely no warranty.
It is mostly in the public domain; some portions are provided under
BSD-style licenses.  See the CREDITS and COPYING files in the
distribution for more information.
* (reverse (cons 1 2))

debugger invoked on a TYPE-ERROR in thread
#<THREAD "main thread" RUNNING {1002A8AF33}>:
  The value 2 is not of type LIST.

Type HELP for debugger help, or (SB-EXT:EXIT) to exit from SBCL.

restarts (invokable by number or by possibly-abbreviated name):
  0: [ABORT] Exit debugger, returning to top level.

(SB-IMPL::LIST-REVERSE* #<unavailable argument>)
0]    
```

Affonso-Gui, 2018-02-08:
clisp also returns error:

```
> (reverse (cons 1 2))

*** - REVERSE: A proper list must not end with 2
```

YoheiKakiuchi, 2018-02-08:
We can check common lisp standard at
http://www.lispworks.com/documentation/HyperSpec/Front/

'reverse' is described at
http://www.lispworks.com/documentation/HyperSpec/Body/f_revers.htm
type-error if sequence is not a proper sequence.

So, SBCL and clisp keep common-lisp standard, but euslisp doesn't.

knorth55, 2018-02-08:
so you mean we need to set our standard?

YoheiKakiuchi, 2018-02-08:
We have to be careful for changing behavior of existing software.
Even if it seems right and easy to fix it.
We have to evaluate profit and bad effect from fixing it.

In this case, I think we can fix it.
some existing programs may change behavior, but not so many.

euslisp manual P.35
reverse _sequence_   /  reverse the order of _sequence_ and returns a new sequence of the same type as _sequence_.
( Not described about improper sequence )

Affonso-Gui, 2018-02-08:
Related to this issue and #258 , I just discovered that the following holds for both sbcl and clisp.
```
> (last (cons 1 2) 0)
2
```
This kind of also explains why implementation of function `last` on sbcl separates cases when `n` is zero but do not return `nil` for all of them.

knorth55, 2018-02-09:
I think we have two options.

1.  follow common lisp, so return improper sequence error.
I prefer this because most of documents on the internet is about common lisp.
It is easier for us to follow common lisp.
2. return something.
I think `(reverse (cons 1 2)) => (cons 2 1)`can be useful.
Actually, I expected to get the value as above.