euslisp/EusLisp #265: (1+ [float-value]) result changes when compiled

mmurooka, 2018-02-22:
`(1+ 5.5)` returns 6.5 when euslisp code is NOT compiled, but it returns 5.5 (same as input argument) when compiled.
Euslisp document says following, so `(1+ 5.5)` is incorrect usage. http://euslisp.github.io/jskeus/jmanual-node39.html (BTW, layout is broken in this page.)
```
1+ integer         [ 関数 ]
integer + 1 を返す。 1+ と 1− の引数は、 integer でなければならない。
```

Even though behavior of incorrect usage might not be defined, the difference between not compiled and compiled is very difficult to debug and troublesome.
`1+` is also used in `incf`, so `(progn (setq x 5.5) (incf x))` also changes after compiling.
https://github.com/euslisp/EusLisp/blob/master/lisp/l/common.l#L118

---
All output of test.
```lisp
leus@cygnus:~$ cat /tmp/test-plus-1.l 
(defun test
  ()
  (format t "(1+ 5) = ~a~%" (1+ 5))
  (format t "(1+ 5.5) = ~a~%" (1+ 5.5))
  )
leus@cygnus:~$ roseus
configuring by "/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0.0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x522a830[16374] --> 0x5689570[32748] top=3d47
irtgl irtglc irtviewer 
EusLisp 9.23(40b86dc 1.0.12) for Linux64 created on cygnus(Sun Feb 18 22:37:58 JST 2018)
roseus ;; loading roseus("3a7f42f") on euslisp((9.23 cygnus Sun Feb 18 22:37:58 JST 2018 40b86dc 1.0.12))
eustf roseus_c_util 
1.irteusgl$ (load "/tmp/test-plus-1.l")
t
2.irteusgl$ (test)
(1+ 5) = 6
(1+ 5.5) = 6.5
nil
3.irteusgl$ (comp:compile-file-if-src-newer "/tmp/test-plus-1.l" "/tmp/")
compiling file: /tmp/test-plus-1.l
gcc -g -c -o /tmp/test-plus-1.o -Dx86_64 -DLinux -Wimplicit -falign-functions=8 -DGCC3  -DGCC  -DTHREADED -DPTHREAD -fpic  -I/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus/include -O2 /tmp/test-plus-1.c; ld -shared -build-id -o /tmp/test-plus-1.so /tmp/test-plus-1.o
"/tmp/test-plus-1.l"
4.irteusgl$ (exit)
leus@cygnus:~$ roseus
configuring by "/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0.0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x5d23830[16374] --> 0x6182570[32748] top=3d47
irtgl irtglc irtviewer 
EusLisp 9.23(40b86dc 1.0.12) for Linux64 created on cygnus(Sun Feb 18 22:37:58 JST 2018)
roseus ;; loading roseus("3a7f42f") on euslisp((9.23 cygnus Sun Feb 18 22:37:58 JST 2018 40b86dc 1.0.12))
eustf roseus_c_util 
1.irteusgl$ (load "/tmp/test-plus-1.so")
#<load-module #X62f5da0 "/tmp/test-plus-1.so">
2.irteusgl$ (test)
(1+ 5) = 6
(1+ 5.5) = 5.5
```

k-okada, 2018-02-28:
1. what is your expected solution (for example #267)
2. any good idea to run test with compiled code?

mmurooka, 2018-02-28:
> any good idea to run test with compiled code?

There are two choices, (i) do all tests after compilation, (ii) do only corner case tests (selected manually) after compilation.
example of (ii) might be like https://github.com/euslisp/jskeus/pull/492 .

mmurooka, 2018-02-28:
> what is your expected solution (for example #267)

+1 for https://github.com/euslisp/EusLisp/pull/267 .

closed 2018-04-04 by k-okada