euslisp/EusLisp #278: Segmentation fault when running the specified function

ericlesaquiles, 2018-05-17:
Just for fun, I tried declaring the following

```
(defun make-dbms (db)
  (list
   #’(lambda (key)
     (cdr (assoc key db)))
   #’(lambda (key val)
     (push (cons key val) db)
       key)
   #’(lambda (key)
     (setf db (delete key db :key #’car))
      key)))

```

and then running 

`1.eus$ setq cities (make-dbms '((belo . br)(paris . fr)))`

getting
```

setq cities (make-dbms '((belo . br)(paris . fr)))
((lambda-closure nil 28834344 0 (key) (cdr (assoc key db))) (la\
mbda-closure nil 28834344 0 (key val) (push (cons ley val) db) \
key) (lambda-closure nil 28834344 0 (key) (setf db (delete key \
db :key #'car)) key))
2.eus$ cities
cities
((lambda-closure nil 28834344 0 (key) (cdr (assoc key db))) (la\
mbda-closure nil 28834344 0 (key val) (push (cons ley val) db) \
key) (lambda-closure nil 28834344 0 (key) (setf db (delete key \
db :key #'car)) key))
```
but then, after running `3.eus$ funcall (car cities) 'belo`, got 

```
funcall (car cities) 'belo
;; Segmentation Fault.
;; in (funcall (car cities) 'belo)
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=1338319920 x=4fc52300 addr=1b7fa28
Fatal:

signal=11 to thread 0,
')
```

whereas the expected return value was "br".

Affonso-Gui, 2018-05-17:
Since Euslisp works with dynamic scope, it does not quite support closures.

```
;; Dynamic scope                                                                                      
(setq a 1)
(let ((a 2)) (defun foo () a))
(foo)
;; 1                                                                                                  
```

Instead of using closures, in Euslisp we use objects to express the combinations of functions (methods) and lexical environment (slots).

The following is what you want:
```
(defclass dbms :slots (db))
(defmethod dbms
  (:get (key) (cdr (assoc key db)))
  (:add (key val) (push (cons key val) db))
  (:delete (key) (setf db (delete key db :key #'car))))

(setq cities (make-instance dbms :db '((belo . br) (paris . fr))))
(send cities :get 'belo)
;; br                                                                                                 
```

You should think of Euslisp as an `object-based` language, rather than an `object-oriented` one.

ericlesaquiles, 2018-05-17:
Thank you, I had overlooked that. I will review the manual once more with more care so as to see this kind of difference quicker in the future

closed 2018-05-17 by ericlesaquiles

Affonso-Gui, 2018-05-17:
Quick add:
What happens in the original code is that closures search for a global definition of `db`, which does not exist.
Defining a global `db` will avoid errors and segmentation faults:
```
(defvar db '((belo . br) (paris . fr)))
```