euslisp/EusLisp #279: too many keyword parameters error when compiling

shintarokkk, 2018-05-24:
When compiling file _fuga.l_ such as

`(defun fuga
  (&key
   k0 k1 k2 k3 k4 k5 k6 k7 k8 k9 k10 k11 k12 k13 k14 k15 k16 k17 k18 k19 k20 k21 k22 k23 k24 k25 k26 k27 k28 k29 k30 k31 k32)
  (pprint "fuga")
  )`
by
`1.irteusgl$ (compile-file "/tmp/fuga.l")`,
error occurs and the message reads
`compiling file: /tmp/fuga.l
Too many keyword parameters>32 (:k0 :k1 :k2 :k3 :k4 :k5 :k6 :k7 :k8 :k9 :k10 :k11 :k12 :k13 :k14 :k15 :k16 :k17 :k18 :k19 :k20 :k21\
 :k22 :k23 :k24 :k25 :k26 :k27 :k28 :k29 :k30 :k31 :k32)err:`.

I thought this error was fixed in https://github.com/euslisp/EusLisp/pull/272, but it still occurs.
What is the cause of this error?

k-okada, 2018-05-24:
please make sure that you have cleaned your source tree, before compile
```
catkin clean
git clean -xfd
catkin -vi b
```
then you'll get
```
k-okada@p51s:~/catkin_ws/ws_euslisp/src/euslisp/lisp$ eus2
configuring by
"/home/k-okada/catkin_ws/ws_euslisp/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;;
string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;;
eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;;
trans ;; comp ;; builtins ;; l/par.l ;; helpsub ;; eushelp ;; fstringdouble
EusLisp 9.23(10158f9) for Linux64 created on p51s(Thu May 24 18:52:59 JST
2018)
1.eus2$ compile-file "/tmp/fuga.l"
compiling file: /tmp/fuga.l
gcc -g -c -Dx86_64 -DLinux -Wimplicit -falign-functions=8 -DGCC3  -DGCC
-DTHREADED -DPTHREAD -fpic
-I/home/k-okada/catkin_ws/ws_euslisp/devel/share/euslisp/jskeus/eus/include
-O2 /tmp/fuga.c; ld -shared -build-id -o /tmp/fuga.so
/tmp/fuga.o/tmp/fuga.c: In function ‘F59fuga’:
/tmp/fuga.c:123:12: warning: left shift count >= width of type
[-Wshift-count-overflow]
  if (n & (1<<32)) goto KEY93;
            ^
ld: cannot find /tmp/fuga.o: No such file or directory

nil
```
if you still have problems, try with `eus0` and if that works, you fave
something wrong on compile

shintarokkk, 2018-05-24:
Thanks, this problem seems to be solved by your direction. What I did was,
・ "catkin clean" at "~/ros/indigo/src" and "~/ros/indigo_parent/src"
・"git clean -xfd" at "~/ros/indigo_parent/src/euslisp/jskeus" and "~/ros/indigo_parent/src/euslisp/Euslisp"
・"catkin build" at "~/ros/indigo/src" (build of package "roseus" failed, but I will write other issue.)
・"catkin build" at "~/ros/indigo_parent/src"
・build Euslisp following README at https://github.com/euslisp/EusLisp
・build jskeus following README at https://github.com/euslisp/jskeus
I feel this procedure was redundant but the error about _compile-file_  was solved.

Thanks.

furushchev, 2018-07-25:
@shintarokkk Do you still have any trouble on this issue, or please close it since this issue is inactive for several months.

closed 2018-07-27 by shintarokkk