euslisp/EusLisp #295: Bug in REMOVE/DELETE sequence with key :end / :count

furushchev, 2018-07-13:
**SBCL**

```lisp
* (remove 1 (vector 1 2 3 2 1) :count 1) ;; -> #(2 3 2 1)
* (delete 2 (vector 1 2 3 2 1) :end 2)   ;; -> #(1 3 2 1)
```

**eus**

```lisp
18.irteusgl$ (remove 1 (vector 1 2 3 2 1) :count 1)  ;; -> #()
9.irteusgl$ (delete 2 (vector 1 2 3 2 1) :end 2)     ;; -> #(1)
```

YoheiKakiuchi, 2018-07-13:
These tests will be better if they contain tests for 'Destructive operation'.

'remove' is not 'Destructive'
(setq a (vector 1 2 3 4))
(remove 1 a) -> #(2 3 4)
(print a) -> #(1 2 3 4)

'delete' is 'Destructive'
(setq b (vector 1 2 3 4))
(delete 1 b) -> #(2 3 4)
(print b) -> #(2 3 4)

furushchev, 2018-07-14:
@YoheiKakiuchi Thanks for the review!
Hyperspec says:

> Sequence may be destroyed and used to construct the result; however, the result might or might not be identical to sequence.

This means theoretically it is not guaranteed that the input sequence is the same as the return value but that the function may or may not break the input value.

You can check this dangerous behavior on `sbcl`:

```lisp
* (setq l (vector 1 2 3 4 5)) ;; -> #(1 2 3 4 5)
* (delete 2 l) ;; -> #(1 3 4 5)
* l ;; -> #(1 3 4 5 5)
```

The return value of `(delete 2 l)` is surely the expected result of this function but after evaluation of this function the input value `l` is completely unexpected and should not be used from that point.
(This could be a problem when we have a chance to migrate euslisp code to common lisp)

As for the specification of `euslisp`, should we define this behavior as the euslisp spec or just a lucky? (I see this usage at some parts in euslisp codes).

furushchev, 2018-07-14:
And just from my curiosity, Where does the difference in performance come from in both cases?:

```lisp
;; Case 1
(setq seq (vector 1 2 3 4 5))
(setq seq (delete 2 seq))

;; Case 2
(setq seq (vector 1 2 3 4 5))
(delete 2 seq)
```

If I understand correctly, the return value of `delete` has the same pointer as that of `seq`, right?
In that case, I can see the overhead of the case 1 is just a `setq` in the second line but I'm not sure the `setq` has such a huge cost...

YoheiKakiuchi, 2018-07-14:
You are right.

Common lisp specification for 'delete' does not concern original list/vector.
It is implementation-dependant behavior.

So, manual of euslisp describes
'''delete modifies seq destructively and does not create a new sequence'''
It does not violate 'Common lisp specification', but add additional specification to 'delete' .
I think it is not good, but for backword compatibility it should not be changed.

> If I understand correctly, the return value of delete has the same pointer as that of seq, right?

Yes, for euslisp.

Is there significant difference in these cases?

```
;; Case 1
(setq seq (vector 1 2 3 4 5))
(setq seq (delete 2 seq))
;; Case 2
(setq seq (vector 1 2 3 4 5))
(delete 2 seq)
```

BTW, 'delete' works like this.
```
(setq a (list 1))
(delete 1 a) ;; -> nil
(print a) ;; -> (1)
```

Then, we should use return value of 'delete', not argument list/vector.
It is the same for 'remove'.
Additionally, argument list/vector should not be used.

furushchev, 2018-07-14:
@YoheiKakiuchi You mean we should follow the common lisp specification, right? What I can do is to add the test case of `delete` argument reuse only for a sequence (not cons) 

https://github.com/euslisp/jskeus/pull/509

Affonso-Gui, 2018-08-10:
Fixed with https://github.com/euslisp/EusLisp/commit/19041083693e2e656697bc2fd0d189e19b43b802.

Can we close this?