euslisp/EusLisp #302: defparameter/subseq issue

ericlesaquiles, 2018-07-18:
```
(defparameter *searched-list*
  '(b b a b b b b b b b a b a b b b a b a b b b a a a a b a a b a a a a a
      a b a b b a b a a b a a a b b a a b a a a a b b a b a b a a a b a b
      b a b a a b b b b b a a a a a b a b b b b b a b a b b a b a b))
```
No problems up to here, but

```
(defparameter *pattern-sublists*
  (remove-duplicates
   (let* ((s *searched-list*) (len (length s)))
     (loop for x from 0 to 8 nconc
           (loop for y from 0 to (- len x)
                 collect (subseq s y (+ y x)))))
   :test #'equal))
```
yields
`/opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 illegal start/end index  in (subseq s y (+ y x))E:`

I realize there's been a discussion on `subseq`-related issues lately, but it seems to me  (perhaps wrongly) to be of a different kind.

obs.: that is using the loop library, which does (pretty much seem to) work.

Affonso-Gui, 2018-07-19:
Addressed on https://github.com/euslisp/EusLisp/issues/298.

Problem is to give `:start` equal to lenght, such as in:
```lisp
(let ((lst (list 1 2 3))) (subseq lst (length lst) (length lst)))
```

This returns `nil` in sbcl and error in eus.

Affonso-Gui, 2018-07-19:
test with:
```lisp
(defparameter *pattern-sublists*
  (remove-duplicates
   (let* ((s *searched-list*) (len (length s)))
     (loop for x from 0 to 8 nconc
           (loop for y from 0 to (- len x)
                 collect (if (zerop x) ;; y = y+x = len
                             nil
                           (subseq s y (+ y x))))))
   :test #'equal))
```

Affonso-Gui, 2018-08-10:
Duplicate of https://github.com/euslisp/EusLisp/issues/298.
Can we close this?

closed 2018-08-10 by k-okada