euslisp/EusLisp #303: [documentation] Separate parameter list from normal documentation

Affonso-Gui, 2018-07-19:
It is great to have the automatic documentation for functions in euslisp:
```lisp
irteusgl$ (defun foo (a b c) (list a b c))
irteusgl$ (documentation 'foo)
(a b c)
```
The problem is that this is overwritten by normal documentation:
```lisp
irteusgl$ (defun bar (a b c) "Bar documentation" (list a b c))
irteusgl$ (documentation 'bar)
"Bar documentation"
```
So in many cases the original purpose of making the function more understandable by adding documentation string has reverse effect, making the function harder to use since we do not know what parameters the it takes.

This could be solved by separating the documentation and parameter list slots in the symbol:
```lisp
irteusgl$ (defun bar (a b c) "Bar documentation" (list a b c))
irteusgl$ (documentation 'bar)
"Bar documentation"
irteusgl$ (function-lambda-list 'bar)
"(a b c)"
```

Sample implementation can be found in https://github.com/Affonso-Gui/EusLisp/tree/builtin-doc.

Should this be implemented? @k-okada

k-okada, 2018-07-25:
is this cl compatible ?

Affonso-Gui, 2018-07-25:
In common lisp we do have separate slots for documentation and argument-list:
```lisp
CL-USER> (describe 'remove)
COMMON-LISP:REMOVE
  [symbol]

REMOVE names a compiled function:
  Lambda-list: (ITEM SEQUENCE &REST ARGS &KEY FROM-END (TEST #'EQL)
                (TEST-NOT NIL) (START 0) (END NIL) (COUNT NIL)
                (KEY NIL))
  Declared type: (...)
  Derived type: (...)
  Documentation:
    Return a copy of SEQUENCE with elements satisfying the test (default is
       EQL) with ITEM removed.
  Known attributes: call, flushable, unsafely-flushable
  Source file: SYS:SRC;CODE;SEQ.LISP
```

Affonso-Gui, 2018-09-08:
I believe the best should have different slots for string documentation and argument list, and update them from the tex docs when available.

Affonso-Gui, 2018-09-10:
Something like:
```lisp

(defun set-doc (name item)
  (when (<= 2 (help::help-item-type item) 3) ;;function or macro
    (let ((param-lst (send item :read-help))
	  (sym (intern (string-upcase name) :lisp)))
      (setf (get sym :lambda-list)
	    (format nil "(~a)" (car param-lst)))
      (setf (get sym :function-documentation)
	    (cadr param-lst)))))

(help '+) ;; create *help-hash*
(maphash #'set-doc help::*help-hash*)
```
Which gives:
```
(documentation '+)
;;  "returns the sum of numbers."

(get '+ :lambda-list)
;; "(&rest numbers)"
```

closed 2019-02-13 by Affonso-Gui