euslisp/EusLisp #304: Upgrading euslisp via apt-get fails on shadow-fixed

furushchev, 2018-07-19:
Just wanted to inform that upgrading euslisp using shadow-fixed fails today:

```bash
Unpacking ros-kinetic-euslisp (9.24.0-0xenial-20180717-075819-0800) over (9.23.0-0xenial-20180222-205130-0800) ...
dpkg: error processing archive /var/cache/apt/archives/ros-kinetic-euslisp_9.24.0-0xenial-20180717-075819-0800_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/kinetic/setup.zsh', which is also in package ros-kinetic-catkin 0.7.11-0xenial-20180222-175501-0800
Errors were encountered while processing:
 /var/cache/apt/archives/ros-kinetic-euslisp_9.24.0-0xenial-20180717-075819-0800_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Maybe invalid configuration for install?
If it does not matter, please close.

k-okada, 2018-07-19:
sorry and that should be fixed with 9.25
https://github.com/tork-a/euslisp-release/pull/15

furushchev, 2018-07-19:
Ah, this is duplicated of tork-a/euslisp-release#14 , sorry.
I got it, thanks anyway.

closed 2018-07-19 by furushchev