euslisp/EusLisp #328: identity function returns wrong result

iory, 2018-09-22:
When we use identity with float-vector, following code returns wrong result.

```
(defun remove-nth (index seq)
  (remove-if #'identity seq :start index :count 1))

(setq seq (float-vector 0 1 2 3 4 5 6 7 8 9))
(print (remove-nth 5 seq))
;; => #f(0.0 1.0 2.0 3.0 4.0)
(setq seq (list 0 1 2 3 4 5 6 7 8 9))
(print (remove-nth 5 seq))
;; => (0 1 2 3 4 6 7 8 9)
```

iory, 2018-09-22:
Another example
```
(setq seq (float-vector 0 1 2 3 4 5 6 7 8 9))
(mapcar #'identity seq)
;; nil
(setq seq (list 0 1 2 3 4 5 6 7 8 9))
(mapcar #'identity seq)
;; (0 1 2 3 4 5 6 7 8 9)
```

furushchev, 2018-09-22:
> (defun remove-nth (index seq)
  (remove-if #'identity seq :start index :count 1))
(setq seq (float-vector 0 1 2 3 4 5 6 7 8 9))
(print (remove-nth 5 seq))
;; => #f(0.0 1.0 2.0 3.0 4.0)

This is a bug reported below:
https://github.com/euslisp/EusLisp/issues/295
and the pull request for the fix was merged into `cl-compatible` branch:
https://github.com/euslisp/EusLisp/pull/296
(though it is not merged into `master`)

> (setq seq (float-vector 0 1 2 3 4 5 6 7 8 9))
(mapcar #'identity seq)

This case, I doubt `mapcar` works with `float-vector` or `sequence`, it is expected to be used only with `list`.

iory, 2018-09-22:
I overlooked the issue and PR.
Thank you for your response.

> This case, I doubt mapcar works with float-vector or sequence, it is expected to be used only with list.

I see. ```mapcar``` supports only list.

closed 2018-09-22 by iory