euslisp/EusLisp #336: string trim is sometimes incorrect

mmurooka, 2018-10-09:
I don't find the reason of this bug yet, just first report.

```
26.irteusgl$ string-left-trim "--dataset=" "--dataset=aaa"
"" ;; <- incorrect !!!
27.irteusgl$ string-left-trim "--dataset=" "--dataset=bbb"
"bbb"
28.irteusgl$ string-left-trim "--dataset=" "--dataset=ccc"
"ccc"
```

Affonso-Gui, 2018-10-09:
This is correct behavior. http://clhs.lisp.se/Body/f_stg_tr.htm

`string-left-trim` takes two arguments, `bag` and `string`, removing characters from string which appear in bag starting from the left. Since the character `a` is also included in "--d**a**t**a**set=", it gets removed and the result is the null string.

Some examples:
```lisp
eus$ (string-left-trim "tsead-" "--dataset=bbb")
"=bbb"
eus$ (string-left-trim (list #\a #\d #\-) "--dataset=bbb")
"taset=bbb"
```
In this case it is probably better to use a simple `subseq` (maybe after a string comparison).

```lisp
(subseq "--dataset=aaa" 10)

(let ((str "--dataset=aaa"))
  (if (string= (subseq str 0 10) "--dataset=")
      (subseq str 10)
    str))

(let ((str "--dataset=aaa")
      (i 0))
  (map nil #'(lambda (a b) (if (= a b) (incf i)))
       str "--dataset=")
  (subseq str i))
```

mmurooka, 2018-10-09:
Thank you! I misunderstood the trim function behavior. Close Issue.

closed 2018-10-09 by mmurooka