euslisp/EusLisp #344: 文字を書けない

mmurooka, 2018-10-31:
@k-okada , 以下のdiffが必要だった記憶があります． @s-noda
https://github.com/mmurooka/euslib/blob/murooka-branch/demo/murooka/wholebody_manipulation_planner/demo_robot_carry_object/euslisp/upstream/glviewsurface-string.l

mmurooka, 2018-10-31:
https://github.com/euslisp/jskeus/pull/530 にPRを出しました．

これって，既存のメソッドをちょっとだけ修正したものかと思っていましたが，
そもそもirtviewerで文字を書く機能はなくて，これは新しくその機能を追加したものという理解であってるでしょうか．

https://github.com/euslisp/EusLisp/blob/master/lisp/xwindow/Xeus.l#L155-L158 をサブクラスでオーバーライドしていることになりますが，中身は全然違うものになっています．

YoheiKakiuchi, 2018-10-31:
オーバーライドもとは、xwindowに文字を表示している。（関数もXのもの）
このPRはOpenGLで文字を表示している。（関数もGLのもの）

今回のPRは、たぶん、巧みにすれば平面に文字を書いて斜めから見られるし、
３次元的な表示位置を指示するのも可能（もとのやつもeus的には３次元位置指定はかのうなはず）

mmurooka, 2018-10-31:
ちなみに，これはむかし@s-noda からもらった秘伝のプログラムだったと思います．もっとはやくPRしておくべきでした．