euslisp/EusLisp #361: Mixed operation in 'list-insert'

Affonso-Gui, 2019-01-18:
In `list-insert`, we have non-destructive operation for null lists or zero position and destructive operation for other cases.

https://github.com/euslisp/EusLisp/blob/master/lisp/l/common.l#L369-L377

Shouldn't these be unified as destructive operation?

Hunter11zolomon, 2019-03-24:
> Shouldn't these be unified as destructive operation?
Can you please elaborate what you mean by unifying the destructive operations?

Affonso-Gui, 2019-03-26:
Sorry, that was not clear enough. I mean that `list-insert` should be destructive for null lists and zero index as well. That is, what now looks like:
```lisp
(setq lst nil)
;; nil
(list-insert 1 0 lst)
;; (1)
lst
;; nil       ;Not Destructive
(setq lst (list 1 2))
;; (1 2)
(list-insert 3 2 lst)
;; (1 2 3)
lst
;; (1 2 3)   ;Destructive
(list-insert 0 0 lst)
;; (0 1 2 3)
lst
;; (1 2 3)   ;Not Destructive
```
Should be:
```lisp
(setq lst nil)
;; nil
(list-insert 1 0 lst)
;; (1)
lst
;; (1)     ;Destructive
(setq lst (list 1 2))
;; (1 2)
(list-insert 3 2 lst)
;; (1 2 3)
lst
;; (1 2 3)   ;Destructive
(list-insert 0 0 lst)
;; (0 1 2 3)
lst
;; (0 1 2 3)   ;Destructive
```

Hunter11zolomon, 2019-03-26:
This looks right to me.
I will try to make a pull request.

Affonso-Gui, 2019-03-27:
Thanks, your help is appreciated.

In case you have not found it yet, the function definition is in https://github.com/euslisp/EusLisp/blob/master/lisp/l/common.l#L369-L377 .

Hunter11zolomon, 2019-03-29:
made a pull request
https://github.com/euslisp/EusLisp/pull/369