euslisp/EusLisp #424: return-from できないときがある

k-okada, 2020-04-08:
```
(defun hoge ()
  (dolist (x (list 0 1 2 3 4))
    (print x)
    (if (eq x 2)
        (return-from hoge nil))))

(hoge)

(defun fuga ()
  (mapcar #'(lambda (x)
              (print x)
              (if (eq x 2)
                  (return-from fuga nil)))
          (list 0 1 2 3 4)))

(fuga)
```
を実行すると以下のようになるんですが、そういうもんでしょうか？
```
0
1
2
0
1
2
Call Stack (max depth: 20):
  0: at (return-from fuga nil)
  1: at (if (eq x 2) (return-from fuga nil))
  2: at (mapcar #'(lambda (x) (print x) (if (eq x 2) (return-from fuga nil))) (list 0 1 2 3 4))
  3: at (mapcar #'(lambda (x) (print x) (if (eq x 2) (return-from fuga nil))) (list 0 1 2 3 4))
  4: at (fuga)
eusgl: ERROR th=0 no such block (x 2) in (return-from fuga nil)E: (exit)
```

Affonso-Gui, 2020-04-08:
Apparently eus creates a block around lambda functions that occludes the previous context blocks.
```lisp
(block :this (mapcar #'(lambda (x) (sys:list-all-blocks)) (list 1 2)))
;; ((nil) (nil))
(mapcar #'(lambda (x) (return 10) x) (list 1 2))
;; (10 10)
```

For reference in sbcl the lambda does not mess with the current blocks:
```lisp
(defun bar () 
           (mapcar #'(lambda (x)
                                   (print x)
                                   (if (eq x 2)
                                       (return-from bar nil)))
                   (list 0 1 2 3 4 5)))
(bar)
; 0 
; 1 
; 2 
; NIL

(mapcar #'(lambda (x) (return 10) x) (list 1 2))
; ==> ERROR
```
https://github.com/euslisp/EusLisp/blob/master/lisp/c/eval.c#L432

PS. using https://github.com/euslisp/EusLisp/pull/403 in the above

k-okada, 2020-04-10:
@Affonso-Gui thaks for comment, did `ci-compatible` solved this problem?

Affonso-Gui, 2020-04-13:
This is not solved in either `cl-compatible` or my developing branch. It is possible to solve this and make the lambda do not create any additional blocks, but again it can have undesired side effects..