euslisp/EusLisp #426: how to draw arrow-object temporally

HiroIshida, 2020-04-24:
I want to draw arrow-object temporally by using `draw-on` method. My attempt was the following
```lisp 
(require "models/arrow-object.l")
(make-irtviewer)
(setq *co* (arrow))
(send *co* :newcoords (make-coords :pos #f(100 100 100)))
(send *co* :draw-on :flush t)
```
but the obtained figure is like the below and it's same as visualization of normal coordinate. 
[image]

I rather want to show arrow object like the below, which is obtained by `(objects *co*)` instead.
[image]

It would be grateful if someone tell me how to display the arrow-object temporally.

k-okada, 2020-04-26:
`:draw-on` is the way to visualize (baseically) coordinates, for example
`(send *pr2* :larm :end-coords :draw-on :flush t)`
--
◉ Kei Okada

Naoki-Hiraoka, 2020-05-29:
I think this code works.
```
(gl::draw-globjects (send *irtviewer* :viewer) (x::draw-things *co*) :draw-origin nil :clear nil)
```

closed 2020-05-30 by HiroIshida