euslisp/EusLisp #429: make-coneで円錐を作りたい

Kanazawanaoaki, 2020-05-18:
```
(setq b1 (make-cylinder 30 2))
(send b1 :translate #f(100 0 0))
(send b1 :rotate (deg2rad -90) :y)
```
で底面の円を作り
```
(setq b2 (make-cone #f(-100 0 0) (send b1 :vertices)))
```
で円錐を作ろうと思ったのですが、
```
(setq b1 (make-cylinder 30 2))
(send b1 :translate #f(100 0 0))
(send b1 :rotate (deg2rad -90) :y)

(setq b2 (make-cone #f(-100 0 0) (send b1 :vertices)))
(objects (list b1 b2))
```
とした.lファイル(https://gist.github.com/Kanazawanaoaki/9ea02bf89b69210402f922e6fe427be5#file-make-cone-test-l)を作成して実行
```
wget https://gist.githubusercontent.com/Kanazawanaoaki/9ea02bf89b69210402f922e6fe427be5/raw/8897d3b04bb1846ce8665d479f391822f575b550/make-cone-test.l
roseus make-cone-test.l
```
すると以下のような図形になってしまいます。
[image]
make-coneをする前に(objects (list b1))をすると上手く行くようです。
```
(setq b1 (make-cylinder 30 2))
(send b1 :translate #f(100 0 0))
(send b1 :rotate (deg2rad -90) :y)

(objects (list b1))

(setq b2 (make-cone #f(-100 0 0) (send b1 :vertices)))

(objects (list b1 b2))
```
```
wget https://gist.githubusercontent.com/Kanazawanaoaki/9ea02bf89b69210402f922e6fe427be5/raw/8897d3b04bb1846ce8665d479f391822f575b550/make-cone-test-2.l
roseus make-cone-test-2.l
```
[image]
`objects`をせずに`:locate` と`:rotate`が`(send b1 :vertices)`に反映されるようにするにはどうすればよいでしょうか。

Kanazawanaoaki, 2020-05-18:
今回の問題とは関係有りませんが、円柱に対して`(send b1 :vertices)`だと上面と下面の点が出てきてしまうので、
```
(setq b1 (make-cylinder 30 2))
(send b1 :translate #f(100 0 0))
(send b1 :rotate (deg2rad -90) :y)

(objects (list b1))

(setq v-list (list (car (send b1 :vertices))))
(dolist (x (cdr (send b1 :vertices))) (if (> (elt x 0) 99) (setq v-list (append v-list (list x)))))

(setq b2 (make-cone #f(-100 0 0) v-list))
(objects (list b1 b2))
```
として一つの面を底面に選ぶのが正しいなと思いました。
```
wget https://gist.githubusercontent.com/Kanazawanaoaki/9ea02bf89b69210402f922e6fe427be5/raw/fff6deaf239b96588d8b7e36cd5506740e4e4dee/make-cone-test-3.l
roseus make-cone-test-3.l
```
[image]

knorth55, 2020-05-18:
こういった変形は遅延評価で実装されています。
これはつまり、`b1`に対して`:translate`や`:rotate`などを行っていますが、評価が毎回行われているわけではありません。
なので、一旦`b1`に対しての`:translate`や`:rotate`に対して評価をしていないと、`(send b1 :vertices)`にこれらの変形が反映されません。
評価するためには、`(send b1 :worldcoords)`をいれたりすると、評価されます。

```lisp
(setq b1 (make-cylinder 30 2))
(send b1 :translate #f(100 0 0))
(send b1 :rotate (deg2rad -90) :y)
(send b1 :worldcoords)

(setq b2 (make-cone #f(-100 0 0) (send b1 :vertices)))
(objects (list b1 b2))
```

Kanazawanaoaki, 2020-05-18:
なるほど、そういうことだったんですね。  
円柱を作れるようになりました。ありがとう御座います。

closed 2020-05-18 by Kanazawanaoaki