euslisp/EusLisp #433: bodyset-linkの:bodiesを変えたい

Kanazawanaoaki, 2020-06-01:
bodyset-linkの:bodiesを変えて中身を変更できるようにしたいと考えています。そのようなことをしているサンプルなどあるでしょうか？  
|potato-model|potato-peel-model|potato-cut-model|
|---|---|---|
|[image]|[image]|[image]  

今は新しくbodyset-linkを同じ位置に定義してすり替えを行っています。  
https://gist.github.com/Kanazawanaoaki/3214f6d7bb23357d48f6c9ecbd04b30b#file-potato-link-test-l

Kanazawanaoaki, 2020-06-02:
一つのbodyset-linkの中身を変えていくのは不可能でも、元のbodyset-linkの情報を受け取って新しいbodyset-linkを作るように出来たらいいかなと思います。
今はすり替える新しいモデルを作る時に楕円球を作るところからやり直しているので、変化前の:bodiesに対して分割や包丁操作時の基準座標をassocした変化後のbodyset-linkを作るように出来たら良いかなと思います。
[image]

k-okada, 2020-06-03:
おそくなりました．
https://gist.github.com/k-okada/8aaff2da42fad9a1374df84a51c8b758

みたいなプログラムが作れるんじゃ何か，という話をしていました．
形状が複雑だと，そうも簡単にはいかないとは思うけど．．．．

[image]

Kanazawanaoaki, 2020-06-03:
ありがとうございます！やってみようと思います。

Kanazawanaoaki, 2020-06-03:
少しいじってみてどういうことをやっているかなんとなく理解しました。  
https://gist.github.com/Kanazawanaoaki/b4a0dbb77e048b68373686eefaac25fa#file-tofu-xyz-l  
[image]
直方体以外の物体も`(dolist (abody geo::bodies)`の`abody`を囲う直方体の幅長さ高さを求める事ができれば、同じように出来るかもしれません。euslispでそのようなバウンディングボックスを出すような事もできるのでしょうか？  そもそも`body-`が出来ない時がある問題を回避する必要も有りますが、、
https://gist.github.com/Kanazawanaoaki/b4a0dbb77e048b68373686eefaac25fa#file-potato-cut-class-test-l
[image]
`geo::bodies`を書き換えた後は`(send *viewer* :draw-objects)`だけでは反映されず`(objects (list *potato*))`とやり直さないといけないみたいですね。

Kanazawanaoaki, 2020-06-03:
バウンディングボックスを出さなくても`(send abody :vertices)`のリス卜の座標の最大値最小値を探索すれば良いと気が付きました。  
https://gist.github.com/Kanazawanaoaki/b4a0dbb77e048b68373686eefaac25fa#file-potato-cut-with-box-approximation-l
[image]
x軸方向で４分割まではそれっぽく出来るようになりました。  
ありがとうございました！

closed 2020-06-03 by Kanazawanaoaki

k-okada, 2020-06-04:
1) euslispでそのようなバウンディングボックスを出すような事もできるのでしょうか？
信じられないですが，「euslisp bouding box」で教えてもらえます．「euslisp make-bounding-box〕だとかなり精度が高くなります．
 https://gist.github.com/k-okada/8aaff2da42fad9a1374df84a51c8b758 も直しましたが
`(setq m (cdar (send abody :csg)))` と等価な情報は` (setq b (make-bounding-box (send abody :vertices)))  (setq m (v- (send b :maxpoint) (send b :minpoint)))` で得られます．

2) geo::bodiesを書き換えた後は(send *viewer* :draw-objects)だけでは反映されず(objects (list *potato*))とやり直さないといけないみたいですね。
これはちょっと残念な直し方しか思いついていませんが， https://gist.github.com/k-okada/8aaff2da42fad9a1374df84a51c8b758 に反映しました．

プログラム的に本当にやりたいのは僕が書いたみたいにきれいに２分割するのではなくて，適当な点？平面？を与えたらそこで分割してくれるプログラムですよね．上で載せてくれたイメージみたいに一方向から切るものだったら，http://euslisp.github.io/jskeus/jmanual-node118.html にある `make-solid-of-revolution`を駆使したら作れるかな，と思ったんだけど，乱切りみたいなのは難しいよねぇ．と．

でも，ちょっとずつ位置を変えて`body-`するのでも良いかもしれない．実際にも思った位置で正確に切れることはないんだから，そこをシミュレートしている，という理屈も成り立つかも．

Kanazawanaoaki, 2020-06-04:
1.も2.も真似をしてみます。ありがとうございます。

>プログラム的に本当にやりたいのは僕が書いたみたいにきれいに２分割するのではなくて，適当な点？平面？を与えたらそこで分割してくれるプログラムですよね．  

そうですね。将来的に他の料理も作れるようにしたいですし、任意の形状の食材に対して任意の切り方（乱切り、輪切り、一口大に切る、この面で切る）が出来るようにするのが理想です。  
カレーを作る分には、玉ねぎ・じゃがいも・トマトは楕円体、人参は円柱か楕円体か`make-solid-of-revolution`で作る回転体に近似して、認識結果から幾何モデルのパラメータを決めてモデルを作って、予め決めた切り方ができればいいかなと思っています。  

切砕したパーツのdissocや色の変更、変更前の位置姿勢を引き継ぐ、事もしたいので今まで作ってきたプログラムと岡田先生が書いてくださったプログラムのいいとこ取りをしていこうと思います。

Kanazawanaoaki, 2020-06-04:
新しくissueをopenして聞くのが良いのかもしれませんが、そもそも`body-`が失敗するのはなぜなのでしょうか？  
https://gist.github.com/Kanazawanaoaki/379ea80e9fe00278fafba96eeeab8258
`(send b1 :vertices)`の頂点が切断面に含まれる時かと思いましたが、球とcubeで試して見ると頂点を含んでも一番上と一番したの頂点を含む時は問題ないようですし、頂点が無くてもエラーになってしまうことも有りますね。
```
1.irteusgl$ zurasu 3                                                                               
47Call Stack (max depth: 20):                                                                        
48  0: at (body- b1 hoge)                                                                            
49  1: at (zurasu 3)                                                                                 
50  2: at #<compiled-code #X5642ed55be18>                                                            
51/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: outer circuit not found    
52 in (body- b1 hoge)   
```
[image]