euslisp/EusLisp #437: objectsの引数の順番で透過の表示結果が異なる

Kanazawanaoaki, 2020-06-18:
https://gist.github.com/Kanazawanaoaki/669f63abda188235a075549ed317390b#file-objects-test-l のようにして、範囲内でランダムな位置姿勢で物体を配置する、のを試していたのですが、`(objects (list *pot* *potato* *nakami*))`と`(objects (list *pot* *nakami* *potato*))`で透過して欲しい部分の表示結果が異なりました。(https://gist.github.com/Kanazawanaoaki/669f63abda188235a075549ed317390b#file-objects-test-l-L307-L309 の部分)  
透過させたい物体をlistの後ろに持って来れば回避出来ますが、これは仕様でしょうか？  
|`(objects (list *pot* *potato* *nakami*))`|`(objects (list *pot* *nakami* *potato*))`|
|---|---|
|[image]|[image]|

YoheiKakiuchi, 2020-06-18:
https://ja.wikibooks.org/wiki/OpenGL%E3%83%97%E3%83%AD%E3%82%B0%E3%83%A9%E3%83%9F%E3%83%B3%E3%82%B0/Transparency
この話ですね。

現在の実装だと、objectsのリストの順に1回だけ描画されるので、透過するものがわかっているなら、それを最後に描画するのが最も簡単な解決策かと思います。

Kanazawanaoaki, 2020-06-18:
ありがとうございます。半透明なオブジェクトの描画は単純な話ではなかったんですね。  

透過させたい時は描画する順番も気をつけようと思います。

Affonso-Gui, 2020-08-06:
@Kanazawanaoaki Is this solved?

Also just for the record `objects` is defined on https://github.com/euslisp/jskeus/blob/master/irteus/irtviewer.l#L584-L594 , so this is technically an euslisp/jskeus issue, not an euslisp/EusLisp one.

Kanazawanaoaki, 2020-08-24:
I'm sorry for being late.

It's not a fundamental solution, but I know how to avoid it.
I will close this issue.

closed 2020-08-24 by Kanazawanaoaki