euslisp/EusLisp #448: :newcoords＋:rotateを繰り返し使うとずれていく

Kanazawanaoaki, 2020-10-06:
`:newcoords`をした後に`:rotate`をする関数を繰り返し使うと、どんどん座標がずれていってしまいます。  

https://gist.github.com/Kanazawanaoaki/2ae3645f4d9f2f391bb1f9d560ab912e#file-arrow-bag-test-l  
のように`newcoords`をした後にそれぞれ45度、90度回転させる関数`(set1)`、`(set2)`を定義すると  
同じ関数繰り返すとずれていき、
```
1.irteusgl$ set1
;; (make-irtviewer) executed
now coords #<coordinates #X559139f31e78  500.0 2000.0 1100.0 / 1.571 0.0 0.785> 
please run (set1) again or (set2) 
nil
2.irteusgl$ set1
now coords #<coordinates #X559139f5ef78  500.0 2000.0 1100.0 / 1.571 0.0 1.571> 
please run (set1) again or (set2) 
nil
```
２つの関数を連続して使うと一回ずつならばうまく行きます。
```
1.irteusgl$ set1
;; (make-irtviewer) executed
now coords #<coordinates #X55d021123e78  500.0 2000.0 1100.0 / 1.571 0.0 0.785> 
please run (set1) again or (set2) 
nil
2.irteusgl$ set2
now coords #<coordinates #X55d0211476b0  500.0 2000.0 1100.0 / 1.571 0.0 1.571> 
nil
```
２つの関数を繰り返し使うとそれぞれでずれていきます。
```
1.irteusgl$ set1
;; (make-irtviewer) executed
now coords #<coordinates #X5582bb968e78  500.0 2000.0 1100.0 / 1.571 0.0 0.785> 
please run (set1) again or (set2) 
nil
2.irteusgl$ set2
now coords #<coordinates #X5582bb98c6b0  500.0 2000.0 1100.0 / 1.571 0.0 1.571> 
nil
3.irteusgl$ set1
now coords #<coordinates #X5582bb602488  500.0 2000.0 1100.0 / 1.571 0.0 1.571> 
please run (set1) again or (set2) 
nil
4.irteusgl$ set2
now coords #<coordinates #X5582bb6128d8  500.0 2000.0 1100.0 / 1.571 0.0 3.142> 
nil
5.irteusgl$ set1
now coords #<coordinates #X5582bb61d028  500.0 2000.0 1100.0 / 1.571 0.0 2.356> 
please run (set1) again or (set2) 
nil
6.irteusgl$ set2
now coords #<coordinates #X5582bb616fb0  500.0 2000.0 1100.0 / 1.571 0.0 -1.571> 
nil
7.irteusgl$ set1
now coords #<coordinates #X5582bb6216e8  500.0 2000.0 1100.0 / 1.571 0.0 3.142> 
please run (set1) again or (set2) 
nil
8.irteusgl$ set2
now coords #<coordinates #X5582bb62e220  500.0 2000.0 1100.0 / 1.571 0.0 -3.798e-15> 
nil
```

解決策としては、回転させる角度が決まっている場合は事前に`:rotate`をした後の座標を`:newcoords`に渡せば良いのかなと思いますが、何が問題で何が起きているのでしょうか？

Kanazawanaoaki, 2020-10-06:
`:rotate`をした後の座標を`:newcoords`に渡すようにしたところ、ずれていくバグは起こりませんでした。  
https://gist.github.com/Kanazawanaoaki/2ae3645f4d9f2f391bb1f9d560ab912e#file-arrow-bag-no-rotate-l
```
1.irteusgl$ set1
;; (make-irtviewer) executed
now coords #<coordinates #X5625f628c1a0  500.0 2000.0 1100.0 / 1.571 0.0 0.785> 
please run (set1) again or (set2) 
nil
2.irteusgl$ set1
now coords #<coordinates #X5625f62a8928  500.0 2000.0 1100.0 / 1.571 0.0 0.785> 
please run (set1) again or (set2) 
nil
3.irteusgl$ set2
now coords #<coordinates #X5625f5f17148  500.0 2000.0 1100.0 / 1.571 0.0 1.571> 
nil
4.irteusgl$ set2
now coords #<coordinates #X5625f5f26950  500.0 2000.0 1100.0 / 1.571 0.0 1.571> 
nil
5.irteusgl$ set1
now coords #<coordinates #X5625f5f1e9a0  500.0 2000.0 1100.0 / 1.571 0.0 0.785> 
please run (set1) again or (set2) 
nil
6.irteusgl$ set2
now coords #<coordinates #X5625f5f2ba78  500.0 2000.0 1100.0 / 1.571 0.0 1.571> 
nil
```

YoheiKakiuchi, 2020-10-06:
jmanual.pdfの p.128 座標系を変更するメソッドに書いてありますが、
:newcoords は代入（コピー）ではなくて、coords（ROS, pos）のインスタンスが同一になります。
https://github.com/euslisp/EusLisp/blob/master/doc/jlatex/jmanual.pdf

それと、#2f((...)) はリードマクロなので、以下の問題と同じで、
set1,set2内のrotationのインスタンスが同一になってしまっているのだろうと想像します。
https://github.com/euslisp/jskeus/issues/583

そういう観点でデバッグしてみてください。
```(sys::address obj)``` すると実際のアドレスが分かります。

Kanazawanaoaki, 2020-10-06:
@YoheiKakiuchi なるほどありがとうございます。  
`(sys::address obj)`でアドレスを見てみると確かに`:newcoods`でセットされているrotが(set1)と(set2)が毎回同じアドレスになっていました。同じ関数を呼ぶたびにその関数で定義されているrotのインスタンスが回転されてしまっていたみたいですね。
https://gist.github.com/Kanazawanaoaki/2ae3645f4d9f2f391bb1f9d560ab912e#file-arrow-bag-address-l
```
1.irteusgl$ set1
;; (make-irtviewer) executed
worldcoords: #<coordinates #X5631a732a7c0  500.0 2000.0 1100.0 / 1.571 0.0 0.785> arrow address: 94771245135112 rot address: 94771257786864 
nil
2.irteusgl$ set1
worldcoords: #<coordinates #X5631a7346d08  500.0 2000.0 1100.0 / 1.571 0.0 1.571> arrow address: 94771245135112 rot address: 94771257786864 
nil
3.irteusgl$ set2
worldcoords: #<coordinates #X5631a6fb2d68  500.0 2000.0 1100.0 / 1.571 0.0 1.571> arrow address: 94771245135112 rot address: 94771257786552 
nil
4.irteusgl$ set2
worldcoords: #<coordinates #X5631a6fc4130  500.0 2000.0 1100.0 / 1.571 0.0 3.142> arrow address: 94771245135112 rot address: 94771257786552 
nil
```
そもそも関数内でリードマクロを使うと、その関数を呼び出すたびに同じインスタンスが呼び出されるみたいなので、今回の問題はその時に値が定義されている値にならないのが問題なのでしょうか。
https://gist.github.com/Kanazawanaoaki/2ae3645f4d9f2f391bb1f9d560ab912e#file-address-test-l
```
 out of func 
94687871314200
94687872225976

 from inside of func 
94687871649832
94687871649832
```

Kanazawanaoaki, 2020-10-06:
根本的な解決には無いっていないのかもしれませんが、`:newcoords`の引数を渡す時に`:copy-worldcoords`を入れて無理やりコピーしてしまってrotのインスタンスが`:rotate`されるのを防ぐことはできました。  
```
  (send *arrow* :newcoords (send (make-coords :pos (float-vector 500.0 2000.0 1100) :rot #2f((0.0 -1.0 0.0) (1.0 0.0 0.0) (0.0 0.0 1.0))) :copy-worldcoords))
```
```
1.irteusgl$ set1
;; (make-irtviewer) executed
worldcoords: #<coordinates #X5584fe2ea160  500.0 2000.0 1100.0 / 1.571 0.0 0.785> arrow address: 94029675230520 rot address: 94029687799680 
nil
2.irteusgl$ set1
worldcoords: #<coordinates #X5584fe305aa8  500.0 2000.0 1100.0 / 1.571 0.0 0.785> arrow address: 94029675230520 rot address: 94029684846664 
nil
3.irteusgl$ set2
worldcoords: #<coordinates #X5584fdf72870  500.0 2000.0 1100.0 / 1.571 0.0 1.571> arrow address: 94029675230520 rot address: 94029684847984 
nil
4.irteusgl$ set2
worldcoords: #<coordinates #X5584fdf823a8  500.0 2000.0 1100.0 / 1.571 0.0 1.571> arrow address: 94029675230520 rot address: 94029687655064 
nil
```
これが解決策なのかなと思います。

closed 2020-10-06 by Kanazawanaoaki

knorth55, 2020-10-06:
ちなみにですが，こういった金沢君のように新しいcoordinateに物体を移動したい場合は，どうやるのがいいのでしょうか？
お勧めはどれなんでしょうか？

YoheiKakiuchi, 2020-10-06:
第一に気を付けることには、defunの中にリードマクロを入れないようにするがあります。

coordinatesクラスを作る時は、
```(make-coords :pos (float-vector x y z) :rpy (float-vector yaw roll pitch))```
がいいように思います。

```:rot```での行列指定の場合は、```(matrix (float-vector 1 0 0) (float-vector 0 1 0) (float-vector 0 0 1))```を使うのだと思いますが、右手系の回転行列以外も与えられるので気をつける必要があります。

また、jmanual.pdfの p.128 座標系を変更するメソッドを書いた経緯としては、
https://github.com/euslisp/EusLisp/blob/master/doc/jlatex/jmanual.pdf

値が直接使われるものを知っておきたいという意図もありました。
値が直接使われるメソッドは要注意で、プライベートにできたらしたいくらいのメソッドかと思います。
なので、極力```:newcoords```は使わずに、```:transform```を使うのが良いのではないかと考えます。