euslisp/EusLisp #453: how can we use rgc?

knorth55, 2020-11-06:
@k-okada 
How can we use rgc which is implemented in 2006?
http://www.jsk.t.u-tokyo.ac.jp/~k-okada/paper/2006_jssst_okada_rgc.pdf

Do we need to include `Makefile.rgc` and build by ourselves?
Some people says euslisp is slow because of gc in macro (I'm not sure), so I want to check it.
https://github.com/euslisp/EusLisp/blob/master/lisp/Makefile.Linux64#L96

k-okada, 2020-11-06:
Yes, you need to recompile to use rgc.
But before you go to this direction , insert the following code and
check when GC runs. If euslisp slows only when it displays "gc ... "
then you can tell gc stops your process, otherwise, you need to
understand what is happening exactly...

(setq sys::*gc-hook* #'(lambda (a b) (format t ";; gc ~A ~A~%" a b)))

--
◉ Kei Okada

knorth55, 2020-11-09:
I checked gc with gc-hook and  I remove all `when` `unless` `dolist`  and stop most of the gc in my control loop.

k-okada, 2020-11-10:
please share the code
--
◉ Kei Okada

knorth55, 2020-11-10:
here, you can find the code.
https://github.com/knorth55/eus_vive/blob/master/euslisp/lib/robot-vr-interface.l

I'm also opening a PR.
https://github.com/jsk-ros-pkg/jsk_control/pull/731

k-okada, 2020-11-10:
> here, you can find the code.
>
> https://github.com/knorth55/eus_vive/blob/master/euslisp/lib/robot-vr-interface.l
>
> I'm also opening a PR.
> jsk-ros-pkg/jsk_control#731

please let me know how to reproduce your problem on my local machine.

knorth55, 2020-11-10:
In order to try that, we need vr controller (vive controller).
I will improve my code to work with other interface like rviz, too, so please wait a week or a month.

I set 10Hz to run the main loop for vr controller, but it wasn't running in 10Hz actually.
It depends on computer spec, so I just want to make my code faster.
Another way is compiling all the codes and make `.so` but I first tried with removing macro method.

k-okada, 2020-11-10:
how about using rosbag to publish vr controller - related topics ,and runs
euslisp code?
--
◉ Kei Okada

knorth55, 2020-11-10:
that's a good idea.
I will correct the rosbag data.