euslisp/EusLisp #456: ロボットから見た目標位置の角度について

mqcmd196, 2020-11-28:
[image]
Eusにおいて、図のようにロボットの世界座標系からみた位置(x, y)、姿勢θ、及び目標点Gの世界座標系からみた位置(x', y')がわかっている時、ロボットからみた目標位置Gの角度φを知るための方法はどのようなもにがあるでしょうか。

Naoki-Hiraoka, 2020-11-28:
以下の2つの手法を思いつきました。
```lisp
(setq x 100)
(setq y 100)
(setq theta pi/2)
(setq x_dash 120)
(setq y_dash 120)

(defun method1 ()
  (setq *robot-coords* (make-coords :pos (float-vector x y 0) :rpy (list theta 0 0)))
  (setq *target-relative-pos* (send *robot-coords* :inverse-transform-vector (float-vector x_dash y_dash 0)))
  (atan2 (elt *target-relative-pos* 1) (elt *target-relative-pos* 0))
  )

(defun method2 ()
  (setq *robot-vec* (float-vector (cos theta) (sin theta) 0))
  (setq *target-vec* (float-vector (- x_dash x) (- y_dash y) 0))
  (vector-angle *robot-vec* *target-vec* #F(0 0 1)))
```

mqcmd196, 2020-11-30:
@Naoki-Hiraoka 
ありがとうございます．`:inverse-transform-vector` という便利なメソッドがあることを初めて知りました．jmanualで確認してみると，ワールド座標系におけるベクトルをローカル座標系における表現に逆変換するというもので，まさに使いたいものでした．
method2はロボットと目標位置のworld極座標の角度を出して，なす角を求めるというやり方でしょうか．

Naoki-Hiraoka, 2020-11-30:
> method2はロボットと目標位置のそれぞれの座標系のなす角を求めるというやり方でしょうか．

座標系は考えずに，ベクトルだけで考えてしまう方法です．現在ロボットが向いている方向のベクトルと，ロボットから目標位置に向かう方向のベクトルを求めて，それらのなす角を求めています．

mqcmd196, 2020-11-30:
理解しました，ありがとうございます．とても助かりました．

closed 2020-11-30 by mqcmd196