euslisp/EusLisp #459: what is the better of macro evaluation?

knorth55, 2021-02-08:
I want to do `funcall` or `apply` like thing with `macro` such as `cond`.
What is the best way to do it?
I do these thing with `eval`.

```lisp
(setq a 1)
(eval (append '(cond) '(((equal a 1) (print "hoge")) (t nil))))
;; "hoge"
```

Affonso-Gui, 2021-02-09:
I cannot think of much besides writing another macro that would do the list operations and act like a funcall or apply. Of course this is only hiding the eval under the defmacro stuff, so I don't think there is anything wrong in going for the `eval` here.

A simplified version could be something like this:
```lisp
(defmacro macro-apply (fn arg-lst) (cons fn arg-lst))
(defmacro macro-funcall (fn &rest args) (cons fn args))

(setq a 1)
(macro-apply cond (((equal a 1) (print "hoge")) (t nil)))
;; "hoge"
```

knorth55, 2021-02-09:
@Affonso-Gui thank you! I will use `eval`.

closed 2021-09-10 by Affonso-Gui