euslisp/EusLisp #468: coordinatesクラスのx-axisメソッドの値がおかしい

ban-masa, 2021-11-05:
https://github.com/euslisp/EusLisp/blob/e33942c6975a6f426b987f584479a3e31b6ff98c/lisp/l/coordinates.l#L51-L53
matrix-rowではなくmatrix-columnが正しいと思われます。

k-okada, 2021-11-05:
なるほど，だから
https://github.com/jsk-ros-pkg/euslib/blame/master/jsk/jskgeo.l#L493-L495
があったのか．

これ，どうしようか．irtgeo.lに追加するのが，正しかったんだろうけど，今まで使った人はいないのかな．
--
◉ Kei Okada

ban-masa, 2021-11-09:
単純にcoordinates.lを修正すればよいかと思っていたのですが、そうではないんでしょうか。

k-okada, 2021-11-11:
この手のバグが難しいのは，間違った仕様のまま使っているプログラムがあるケースです．
しかも，かなり根本的なところなので，気が付かない間にこれが変更されていてプログラムが動かくなった場合，なかなか気が付かないかと思います．

今どこで使われているか，というのは例えば以下のように探せるんだけど，これで大丈夫かどうかもちょっと心配．
https://github.com/search?q=org%3Ajsk-ros-pkg+%22%3Ax-axis%22&type=Code
https://github.com/search?q=org%3Astart-jsk+%22%3Ax-axis%22&type=code

例えば今回 @ban-masa はこのバグに気が付いた，ということなわけど，同じように気が付いたど，どこで使っているか第三者が調べられれば，逆に，ここで僕たちが調べて，使っている人に注意喚起できるのでよいんだけど．．．

で，現実的にはあまりそういうことはないけど，irtgeo.l で直すばあいは euslisp だけを使っている人には影響はないので，若干気が楽に自分で思いこめる，ということでした．なので，直すならcoordinates.l でいいんだけど，副作用の影響範囲をどう見積もるか，というのが悩みどころです．

まぁ，使っている人パッと見ていないから，エイ，と変えてしまうのでもいいんだけど．．

ban-masa, 2021-11-11:
少なくとも今後axis系メソッドを使う人のためにもどちらかでは直しておきたいと思っています。
個人的には以下の理由でcoordinates.lを修正しても良いのではないかと考えています。

1. irteusを使っている人のほうがかなり多いのではないかということ。
2. x-axisを使っている人もバグに気づかず使っているのではないかということ（意図した挙動をしていないことに気づいていない可能性）
3. バグに気づいた人はx-axisメソッドを使わないか自分で再定義しているであろうこと。
4. 修正で実際に影響を受けるのはバグに気づいた上でx-axisメソッドの返り値を何かに使用している人であるが、回転行列の行ベクトルは使いみちに乏しい気がするので使ってる人はいないのではないかということ。

YUKINA-3252, 2022-12-09:
@k-okada 
こちらの問題なのですが、私もaxis系メソッドで得られる値を使いたいと思い、値が誤っていることに気づき、このissueで原因と経緯を知りました。私自身はメソッドをoverrideしました。
（このissueが立てられてから今まで他に言及されている方がいらっしゃらなかったので、新しくこのメソッドを使おうとする人があまりおらず、むしろ直すことで岡田先生がおっしゃるように間違った仕様のまま使っているプログラムに支障が出る影響のほうが大きい可能性のほうが高いと思いましたが一応の報告です。）

Affonso-Gui, 2022-12-09:
検索してみたけどヒットしたのは
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus_tutorials/euslisp/template_grasp_samples.l#L261
だけでした。`:y-axis`と`:z-axis`はヒット件数ゼロ。少なくとも他のベース関数に使われていることはなさそうです。

で、gitlabでも検索しようとしていたのですが、Advanced Searchができませんでした、、ライセンスはあると思うけど手動でenableしないといけない模様?
https://gitlab.jsk.imi.i.u-tokyo.ac.jp/help/user/search/advanced_search.md
https://docs.gitlab.com/ee/integration/advanced_search/elasticsearch.html#enable-advanced-search

Affonso-Gui, 2022-12-10:
なるほど、euslibにありましたか。

gitlabはアカデミックライセンスで使えることはないですか。
https://about.gitlab.com/solutions/education/
https://about.gitlab.com/handbook/marketing/community-relations/community-programs/education-program/

githubもbetaではliteral searchも追加しているらしいので、とりあえずwaitlistに入りました。
https://github.com/features/code-search

Affonso-Gui, 2023-01-10:
github search betaに入れました。
`euslib`, `jsk_control`, `jsk-ros-pkg-unreleased`あたりに結構ありますね。