euslisp/EusLisp #475: copy-tree does not copy float-vector

knorth55, 2021-12-14:
I know `copy-tree` is for `list`, but `copy-tree` returns no error with `float-vector` and does not copy the vector.
It is better to warn `copy-tree` with float-vector.

Affonso-Gui, 2021-12-15:
The implementation explicitly returns the non-cloned value when the argument is not a list, so this doesn't seem like some sort of upredicted side effect (i.e. designed for lists but that "accidentally" could handle other sequence types).
https://github.com/euslisp/EusLisp/blob/master/lisp/c/lists.c#L236

I am not sure if we should solve this by changing the implementation to also copy non-lists; add a warning; or just add more specific jmanual documentation. However, this is very similar to https://github.com/euslisp/EusLisp/issues/361 , which promises destructive results but only delivers that on some cases.

knorth55, 2021-12-15:
I think it is better to keep old behavior for keep backward compatibility, but we can warn the misusage or non-copied behavior.
so I open a PR #476

Affonso-Gui, 2021-12-15:
(just for future reference I would like to add that in this case `copy-seq` does what you want)
```lisp
irteusgl> (setq a (float-vector 1 2 3))
#f(1.0 2.0 3.0)
irteusgl> (setq b (copy-seq a))
#f(1.0 2.0 3.0)
irteusgl> (setf (elt b 0) 5)
5
irteusgl> a
#f(1.0 2.0 3.0)
irteusgl> b
#f(5.0 2.0 3.0)

irteusgl> (sys:address a)
94797007551080
irteusgl> (sys:address b)
94797007912976
```

knorth55, 2021-12-15:
`copy-seq`, which I use it for `float-vector` normally.
I misuse `copy-tree` for `float-vector`, which returns non-cloned `float-vector`, and I spent several times for debugging it.